ControlSystemStudio/cs-studio #1603: CS-Studio Common Training (Lead: Kunal - Reviewers: Will)

berryma4, 2016-02-09:


willrogers, 2016-02-10:
Notes on 0 CS-Studio an introduction:

* Toolbar | New Databrowser Plot _doesn't_ open the Databrowser perspective, although I think it should (page 17)
* Views and editors _can't_ be dragged between different windows.  I wish they could.  See #1329. (page 21)

willrogers, 2016-02-10:
Notes on **1 CS-Studio - probe**

* CSS menu is now CS-Studio (page 2, 11)

willrogers, 2016-02-10:
Notes on **2 CS-Studio - DataBrowser**

* None of the methods of opening the databrowser also open the perspective (page 3)
* The databrowser plot toolbar has changed (page 7)

willrogers, 2016-02-10:
Notes on **2 CS-Studio - BOY**

* It's too hard to get into OPI Runtime perspective with the example OPIs open (page 5)

willrogers, 2016-02-10:
That's all the stuff that is particularly relevant to us at the moment.  I realise lots of things have changed since you last updated these so I'm not being critical.

shroffk, 2016-02-16:
TODO:
- [ ] github repo with the training resources from dropbox.
- [ ] update the presentations with the previously commented changes.

shroffk, 2017-11-14:
TODO:
Create a cs-studio/phoebus google account which will come with 15GB of storage which will be enough.

https://github.com/shroffk/cs-studio-training
This repo needs to be moved to the main cs-studio group

claudio-rosati, 2017-11-15:
Created.