ControlSystemStudio/cs-studio #1608: Symbol Widgets Transparency

jbobnar, 2016-02-10:
Request by Nadine:

Transparent property is available already for some widgets, like:
- Shape widgets -  Arc, Polyline, Polygon, Rectangle, Rounded Rectangle and Ellipse
- Label widget
- Text Update widget
- Scaled widgets - Thermometer, Gauge, Knob, Progress Bar, Scaled Slider, Tank and Meter
- XY Graph widget
- Menu button widget
- Grouping Container widget.

But it is not for Image widget, Image Boolean Indicator widget, and Boolean/Multistate Symbol Image widgets (monitor and control), which all support svg format as well as gif and png formats.

closed 2016-02-10 by berryma4

dmichel76, 2016-03-22:
Actually it seems the transparency behaviour for the Image widget (when using a file format that supports transparency like png) has changed since 4.1.0

Before 4.1.0, the transparent part of the image would act as transparent on the OPI (which would be my expectation as a user) so that other widgets below it can be seen. Since 4.1.0, the transparent pat of the image does not show widgets below and instead appear filled with the colour defined in the "Background Color" field of the Image widget (before, that field didn't have any effect)

Now, I don't which behaviour is the preferred one for most users, but I personally prefer the old behaviour since I can _colorize_ parts of my image by using other Graphics widgets sitting below the Image widget, colour which can then be controlled by rules/pv changes/scripts/etc. For instance, if I have a complex drawing of a given device, I can programatically color parts of it, reacting to PV values.

matthieulambertiter, 2016-03-29:
It is working in the version 4.3.0