ControlSystemStudio/cs-studio #1619: OPIEditor offers to save outline and properties view

jbobnar, 2016-02-11:
If done in a specific way, eclipse marks some views as dirty and offers to save them, when you close CSS. It's a cosmetic bug, but it is confusing the first time you see it.

There is 100 % reproducible way to achieve this state:
1. Close OPIEditor perspective, close outline and properties view
2. Open an opi file in edit mode
3. Modify the contents (add widget, move widget, change property)
4. Open OPI Editor perspective
The outline and properties view will be shown and they will be marked as dirty, because their owner (opi editor) is dirty.

The dirty outline view is a bug in OpiBuilder, but the dirty properties is a resurfacing bug in Eclipse (https://bugs.eclipse.org/bugs/show_bug.cgi?id=372799).

willrogers, 2016-02-11:
This looks like a bug to me.

kasemir, 2016-02-15:
Thank you for finding that `getSaveablePart() { return null; }` fix!
The same issue is in the new display builder and may also be in the data browser.

kasemir, 2016-10-12:
Closing, since we've long merged the fix for this.

closed 2016-10-12 by kasemir