ControlSystemStudio/cs-studio #1622: code format xml

berryma4, 2016-02-16:
It was brought up in #1621, that we have a checkstyle xml in cs-studio/build but not a code format xml.  Kay mentioned he has one:

https://github.com/ControlSystemStudio/org.csstudio.sns/blob/master/plugins/org.csstudio.sns.product/code_format.xml

maybe we could move this to cs-studio/build

And, it was also mentioned that we don't document/advertise this well enough.

willrogers, 2016-02-16:
I bet Kay's style puts opening braces on the next line after function declarations etc.  We could never agree on that.

But maybe I am prepared to give in on this for the sake of consistency...

kasemir, 2016-02-16:
Of course it balances the braces such that opening and closing brace are on the same indentation level.
It's the right thing to do. It's the GNU style https://en.wikipedia.org/wiki/Indent_style#GNU_style, ideal for free software. The K&R style was only useful for saving one line of your 20 line ASCII terminal, but modern displays use less energy plus give you more lines.

OK, this topic sometimes tends to start flame wars, but there's really no need if everybody just puts matching indentation to their braces ;-)

willrogers, 2016-02-17:
There's no need for flame wars once everyone understands what the correct answer is...

Bascially I don't mind if Kay's format is added to the repo.  There's still a problem of discoverability though.

Should some basic guidelines on contributing be added to `README.md`?

closed 2018-06-12 by berryma4