ControlSystemStudio/cs-studio #1631: RAP Features

jbobnar, 2016-02-26:
RAP Features should provide only the rap stuff and not the plugins, which are already included in other non-rap features. After that the rap products should be made "more feature based."

jbobnar, 2016-03-01:
Done. All three war products are now feature based. 

Checkout master from https://github.com/ControlSystemStudio/org.csstudio.rap.product, then execute mvn clean verify. Providing that all necessary plugins/features already exist in the CSS p2 repositories, you will end up with three war files, which are conveniently copied to org.csstudio.rap.product/wars folder.

- The databrowser product contains only the BEAUtY reader plugin. If any other plugin is needed, it has to be added as an additional feature to the org.csstudio.trends.databrowser2.rap.product 
- Boy product contains the core widgets and the databrowser widget (with BEAUtY). The alarmtable.opiwidget is not included, because it increases the size of war for 25%. The ..opibuilder.widgets.extra plugin is not included because it is not RAP compatible.

berryma4, 2016-03-01:
That is great!  Thank you!

jbobnar, 2016-03-03:
Found another issue related to features. 

When pvmanager was replaced with diirt, the pvmanager feature was removed from the cs-studio/core/. However, not all pvmanager plugins are obsolete - the UI plugins are still needed. Until recently they were still present in the RAP feature (which didn't make sense, since RAP does not use them), but it was the RAP feature that actually made the entire build possible. When I restructured RAP features, the plugins were no longer available in any repository. They need to be added back.

closed 2016-03-08 by jbobnar