ControlSystemStudio/cs-studio #1639: Remove unitconversion plugin from build

nickbattam, 2016-03-03:
The ```gov.bnl.unitconversion``` plugin was removed from cs-studio-thirdparty on Feb 19th.

This is still required by the ```org.csstudio.service.unitconversion``` and ```org.csstudio.unitconversion.ui``` plugins which are built but not included in any feature.

Is it ok to remove these modules from the appunorganised pom? The hanging dependency is causing our build to fail.

I notice there are already a lot of commented-out modules in this pom. Would anyone object to me deleting those lines?

kasemir, 2016-03-03:
Not 100% sure because Kunal would know most about any BNL code.
But yes, I see the same build error, so if you can fix it, I'm all for that.

berryma4, 2016-03-03:
I merged the removing from the build, @shroffk can add it back when it's
fixed.

shroffk, 2016-03-11:
Yes,
This is fine.