ControlSystemStudio/cs-studio #1644: PVTable breaks the build

jbobnar, 2016-03-04:
@charlesHenryPATARD 
The latest changes in pvtable broke the build. There are some references to classes Configuration and Measure, but the classes are nowhere to be found. Probably you forgot to add them to git.

I would also like to ask you to reduce the version number on the org.eclipse.core.runtime to 3.11.0, or remove the version number completely. ITER is locked to eclipse mars (without service pack), and it is too much hassle to upgrade to SR1. Thank you!

jbobnar, 2016-03-04:
Several action classes are also missing.

berryma4, 2016-03-04:
#1587

berryma4, 2016-03-04:
See Jenkins for build errors:
https://openepics.ci.cloudbees.com/job/cs-studio-applications-display-master/44

nickbattam, 2016-03-07:
This merge added 705 new checkstyle warnings.
A quick look suggests they are all from tabs and trailing whitespaces.

berryma4, 2016-03-08:
this merge (#1647) added more checkstyle errors

jbobnar, 2016-03-08:
I get the following error when adding the pvtable to the product

```
!ENTRY org.csstudio.display.pvtable 4 0 2016-03-08 16:27:32.816
!MESSAGE FrameworkEvent ERROR
!STACK 0
org.osgi.framework.BundleException: Could not resolve module: org.csstudio.display.pvtable [654]
  Bundle was not resolved because of a uses contraint violation.
  org.osgi.service.resolver.ResolutionException: Uses constraint violation. Unable to resolve resource org.csstudio.display.pvtable [osgi.identity; osgi.identity="org.csstudio.display.pvtable"; type="osgi.bundle"; version:Version="4.3.1.qualifier"; singleton:="true"] because it is exposed to package 'org.apache.poi.ss.usermodel' from resources org.apache.poi [osgi.identity; osgi.identity="org.apache.poi"; type="osgi.bundle"; version:Version="3.13.0"] and org.apache.poi.ooxml [osgi.identity; osgi.identity="org.apache.poi.ooxml"; type="osgi.bundle"; version:Version="3.13.0"] via two dependency chains.

Chain 1:
  org.csstudio.display.pvtable [osgi.identity; osgi.identity="org.csstudio.display.pvtable"; type="osgi.bundle"; version:Version="4.3.1.qualifier"; singleton:="true"]
    require: (&(osgi.wiring.bundle=org.apache.poi)(bundle-version>=3.13.0))
     |
    provide: osgi.wiring.bundle: org.apache.poi
  org.apache.poi [osgi.identity; osgi.identity="org.apache.poi"; type="osgi.bundle"; version:Version="3.13.0"]

Chain 2:
  org.csstudio.display.pvtable [osgi.identity; osgi.identity="org.csstudio.display.pvtable"; type="osgi.bundle"; version:Version="4.3.1.qualifier"; singleton:="true"]
    import: (osgi.wiring.package=org.apache.poi.xssf.usermodel)
     |
    export: osgi.wiring.package: org.apache.poi.xssf.usermodel; uses:=org.apache.poi.ss.usermodel
    export: osgi.wiring.package=org.apache.poi.ss.usermodel
  org.apache.poi.ooxml [osgi.identity; osgi.identity="org.apache.poi.ooxml"; type="osgi.bundle"; version:Version="3.13.0"]
	at org.eclipse.osgi.container.Module.start(Module.java:434)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1582)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1561)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1533)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1476)
	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340)
```

It can be solved by replacing the package import by the plugin import. Is there a reason, why the table imports the package org.apache.poi.xssf.usermodel instead of the whole plugin?

berryma4, 2016-03-08:
I have a fix at https://github.com/ControlSystemStudio/cs-studio/commit/a3f4a485b4d24dddc55b8c2119778de9a7c5dcee
but, I still have the test failing with a 
```
Could not start test bundle: org.csstudio.display.pvtable
org.osgi.framework.BundleException: Error loading bundle activator.
        at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleCon
textImpl.java:711)
        at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(Equino
xBundle.java:941)
        at org.eclipse.osgi.internal.framework.EquinoxBundle$EquinoxModule.start
Worker(EquinoxBundle.java:318)
        at org.eclipse.osgi.container.Module.doStart(Module.java:571)
        at org.eclipse.osgi.container.Module.start(Module.java:439)
        at org.eclipse.osgi.internal.framework.EquinoxBundle.start(EquinoxBundle
.java:392)
        at org.eclipse.osgi.internal.framework.EquinoxBundle.start(EquinoxBundle
.java:411)
        at org.eclipse.tycho.surefire.osgibooter.OsgiSurefireBooter.getBundleCla
ssLoader(OsgiSurefireBooter.java:165)
        at org.eclipse.tycho.surefire.osgibooter.OsgiSurefireBooter.createCombin
edClassLoader(OsgiSurefireBooter.java:98)
        at org.eclipse.tycho.surefire.osgibooter.OsgiSurefireBooter.run(OsgiSure
fireBooter.java:91)
        at org.eclipse.tycho.surefire.osgibooter.HeadlessTestApplication.run(Hea
dlessTestApplication.java:21)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.
java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAcces
sorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at org.eclipse.equinox.internal.app.EclipseAppContainer.callMethodWithEx
ception(EclipseAppContainer.java:587)
        at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandl
e.java:198)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runAppli
cation(EclipseAppLauncher.java:134)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(Ec
lipseAppLauncher.java:104)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.ja
va:380)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.ja
va:235)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.
java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAcces
sorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:669)
        at org.eclipse.equinox.launcher.Main.basicRun(Main.java:608)
        at org.eclipse.equinox.launcher.Main.run(Main.java:1515)
        at org.eclipse.equinox.launcher.Main.main(Main.java:1488)
Caused by: java.lang.ClassNotFoundException: org.csstudio.display.pvtable.Plugin
 cannot be found by org.csstudio.display.pvtable_4.3.1.201603081544
        at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(Bundl
eLoader.java:439)
        at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.
java:352)
        at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.
java:344)
        at org.eclipse.osgi.internal.framework.BundleContextImpl.loadBundleActiv
ator(BundleContextImpl.java:754)
        at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleCon
textImpl.java:706)
        ... 28 more
```

charlesHenryPATARD, 2016-03-08:
Yes I know,
There is a problem with poi maven bundles. Package org.apache.poi.ss.usermodel is provided by org.apache.poi and org.apache.poi.ooxml. I import those plugins as source code and delete in MANIFEST.MF the import package org.apache.poi.ss.usermodel from org.apache.poi.ooxml.

The two plugins have the same name !! When the workspace is build there is no problem but in runtime not.

berryma4, 2016-03-08:
I've had this explained to me in the past on a similar situation:

Caused by: java.lang.NoClassDefFoundError: org/eclipse/core/runtime/Plugin

I was guessing this is an import issue, but that looks fine:

Import-Package: javax.swing,org.eclipse.core.runtime;common=split;vers
 ion="3.4",org.eclipse.core.runtime.preferences;version="3.3",org.osgi.fra
 mework;version="1.6"

> The explanation for this error is quite involved, so I’ll give you a tl;dr. You need to add the following to your instructions for the bundle-plugin:

	<Import-Package>*;ui.workbench=!;common=!;registry=!;texteditor=!;text=!</Import-Package>

> Now the explanation, fasten your seatbelt.

> Eclipse suffers massively from split packages as a consequence of using Require-Bundle for all their dependencies rather than Import-Package, which everybody knows is the right way to describe dependencies in OSGi. One example of this is the org.eclipse.core.runtime package, which is split across the two bundles org.eclipse.equinox.common and org.eclipse.equinox.registry. They both export part of the package, but they do so with a mandatory attribute of “common” and “registry” respectively.

> Fortunately there is a bundle that aggregates the split package: this is the org.eclipse.core.runtime bundle (here it’s easy to get confused because the bundle has the same ID as the package we are talking about). The core runtime bundle exports the *whole* package and it does so without any mandatory attributes. This is good because if you do a plain old “Import-Package: org.eclipse.core.runtime” then you get the whole package and everything works fine.

> Unfortunately bnd (on which the maven-bundle-plugin is based) does something which is… err… of debatable merit. When it sees you build against a bundle which has a mandatory attribute on its export, it assumes that you must want to import that package and therefore it adds the attribute on your Import-Package. You can see this in your manifest with the “common=split” after the package name.

> As a result, OSGi wires you to the partial package exported by the org.eclipse.equinox.common bundle, which doesn’t contain the Plugin class and therefore you get the CNFE.

> The solution I gave at the top simply overrides bnd’s default behaviour and tells it not to add the mandatory attribute. It also includes a bunch of other rules that I have found are necessary due to other splits in the Eclipse/Equinox bundles.

> I hope that helps and made some kind of sense.

berryma4, 2016-03-08:
So, in the end, adding org.apache.poi and org.apache.poi.ooxml as required bundles should fix the problem.
No?

charlesHenryPATARD, 2016-03-08:
Unfortunately not. In Java standard development two packages in two different project with the same name  it's ok but in RCP it's forbidden. How did you create the plugins (org.apache.poi and org.apache.poi.ooxml) @berryma4 ? It sould be better if the two jars can be merged in one plugin.

berryma4, 2016-03-08:
I'll need to fix the import package, like above.
(I'm talking about in the maven-osgi-bundles repo)
I am surprised that adding both as a bundle require didn't fix the issue.
But, a broken bundle is a broken bundle.

berryma4, 2016-03-08:
with the bundle added to required bundles, ... I don't get this error.

berryma4, 2016-03-08:
Ah, I do if I click export to Excel

berryma4, 2016-03-08:
I'm going to suggest using the servicemix version as stated in:
https://bz.apache.org/bugzilla/show_bug.cgi?id=57857
Although, I'm running into the same issue they stated with org.etsi.uri.x01903.v14
```
    -> Bundle-SymbolicName: org.apache.servicemix.bundles.poi; bundle-version="3.13.0.1"
       org.apache.servicemix.bundles.poi [465]
         Unresolved requirement: Import-Package: com.sun.javadoc; resolution:="optional"
         Unresolved requirement: Import-Package: com.sun.tools.javadoc; resolution:="optional"
         Unresolved requirement: Import-Package: org.apache.crimson.jaxp; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.asn1; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.asn1.cmp; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.asn1.nist; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.asn1.ocsp; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.asn1.x500; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.asn1.x509; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.cert; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.cert.jcajce; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.cert.ocsp; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.cms; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.cms.bc; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.operator; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.operator.bc; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.tsp; resolution:="optional"
         Unresolved requirement: Import-Package: org.bouncycastle.util; resolution:="optional"
         Unresolved requirement: Import-Package: org.etsi.uri.x01903.v14
```

I'm going to down grade the version, and hope they supply a bundle soon.

kasemir, 2016-03-09:
If the libraries for creating binary Excel files are such a snakepit, would it make sense to instead create ASCI-type *.cvs files?
They can be imported by Excel (and gnumeric, and open office, and python/numpy, gnuplot, ..).

OK, binary Excel files can include settings for fonts, colors, .., but I doubt that's actually required for this export-to-Excel functionality. In fact it might even be better to not mention Excel at all, and just call it the 'spreadsheet' or '*.csv' export.

charlesHenryPATARD, 2016-03-09:
Can we create a platform plugin as we have done before ? This solution work well. We add inside the poi and poi.ooxml jars and that it.
What do you think about this solution ? We can change in the future for maven osgi bundles.

berryma4, 2016-03-09:
I've added the servicemix version to the maven-osgi-bundles repo:
https://github.com/ControlSystemStudio/maven-osgi-bundles/commit/7b43bfd631f12e45f5365ccf774b571c7259d16d
I had to reduce a few of the exports, so I appended the version name with ".css1".
When apache poi gets their bundle working, we can switch back.

berryma4, 2016-03-09:
@jbobnar, @kasemir does this work for you now?
Everything seems to be working for me in both the IDE and tycho.
I'm testing by adding a few pvs to a pvtable, then exporting to excel using the button in the coolbar.

kasemir, 2016-03-09:
My "nightly" build works again

closed 2016-03-09 by berryma4

jbobnar, 2016-03-10:
It works for me as well.