ControlSystemStudio/cs-studio #1651: Mixed Java versions

nickbattam, 2016-03-09:
CSStudio is now a 'Java 8' product. 

The project's plugins are explicitly specifying a mix of versions of Java in the `MANIFEST.MF` files for the RequiredExecutionEnvironment: 
- J2SE-1.5 (51 plugins), 
- JavaSE-1.6 (287 plugins), 
- JavaSE-1.7 (175 plugins)
- JavaSE-1.8 (132 plugins).

The only direct problem I have seen this cause are a consequence of changes to the `@Override` annotation between 1.5 and 1.6 - it changed from being illegal on interface implementations to optional. This results in some checkstyle warnings and an inconsistent use of the annotation across the codebase. 

There may be other more subtle problems buried away.

Can anyone see any problems in updating Bundle-RequiredExecutionEnvironment to be consistent across the project? If not a full update, bumping up the Java5 plugins to require 1.6 (at least) would seem like a good idea as Java5 went end-of-life in 2009 and retaining support for it is now producing warnings.

kasemir, 2016-03-09:
Yes, we now need Java 8: Some code already uses lambdas and 'Instant', and we're planning to update the VTypes to Instant in the near future. Doesn't mean that we have to update every MANIFEST to require Java 8. If the code in the plugin is satisfied with just Java 5, there's no operational requirement to change that.
But I agree that it would be good to change them, because that way we remove some IDE warning about the current IDE's default runtime (java 8) not being a perfect match for the plugin's requested execution environment.
Also helps consolidate the different `@Override` handling on interfaces - which also means some older code may need to be updated.
Finally, whenever we change some code in older plugins, having them already list Java 8 in the MANIFEST may be a mental nudge to use Lambda, Instant, streams as appropriate.

So if you have spare cycles to update the MANIFEST files and fix the few `@Override` errors on interfaces, go for it.

jbobnar, 2016-03-17:
@nickbattam 
Some of the plugins have local .settings folder, which might specify the compiler settings. When you change the RequiredExecutionEnvironment, you should also remove the compiler settings or change them accordingly. Eclipse uses those settings to compile the workspace. If the compiler is set to 1.5 and there are Override tags on the interface methods, eclipse will complain.

kasemir, 2016-03-17:
We're trying to exclude the .settings from the GIT repository, see top-level .gitignore:

```
# ignore the .settings folder
# should it be changed to all */.settings/* ?
.settings/
```

The IDE still tends to create such .settings folders, which you then need to delete or update as you change the  RequiredExecutionEnvironment. In my case, just deleting them solved the compiler errors in the IDE after the recent RequiredExecutionEnvironment updates.

jbobnar, 2016-03-17:
There are some relics like this: https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/opibuilder/opibuilder-plugins/org.csstudio.swt.widgets.rcp/.settings/org.eclipse.jdt.core.prefs
There are a few others which are still under git control and should be removed.

willrogers, 2016-04-06:
We have decided to remove all `.project`, `classpath` and `.settings` files from Git.

See #1723.

closed 2016-04-06 by willrogers