ControlSystemStudio/cs-studio #1668: Proliferation of logging plugins

nickbattam, 2016-03-16:
A quick look in the m2/p2 directories created during a build shows a proliferation of different logging frameworks being used and multiple versions of each (there may be others): 

jul: (core library)
slf4j: 1.5.6, 1.5.10, 1.5.11, 1.6.4, 1.6.6, 1.7.2, 1.7.5, 1.7.7
log4j: 1.2.12, 1.2.13, 1.2.14, 1.2.16, 1.2.17, 1.6.4, 1.7.5, 1.7.7
apache-commons-logging: 1.0, 1.0.3, 1.0.4, 1.1, 1.1.1
(maven) doxia-logging: 1.1, 1.2, 1.4, 1.6

Issue #1060 identified a number plugins that were still using slf (mainly DESY plugins).

To simplify dependency management, and bring more consistency to the way logging is implemented across the source it might be worth moving all the cs-studio (and by implication cs-studio-thirdparty, maven-osgi-bundles and diirt) source to use a single version of a single logging tool.

kasemir, 2016-03-16:
In principle, we've agreed some time ago to use java.util.logging.
We did that exactly to prevent a growth of logging mechanisms.
Even though each might have some minor advantage, java.util.logging is "there", it handles the basics of logging.

Though nobody has to use that in their product, we added the org.csstudio.logging and logging.ui plugins to support JMS logging and preferences for configuring java.util.logging. Kunal recently added the more detailed log configuration tree UI.

The biggest downside of jul was potentially creating expensive message strings for a disabled log level. The java 8 update for message suppliers fixes that.

So in csstudio code, remains of using other logging libs should just be a matter of insifficient time to remove it from existing code. We can't do anything about 3rd party code pulling in new log libraries, though.

kasemir, 2016-04-05:
Removed SLF4J from maven-osgi-bundles, which was only used by DESY plugins.

Remaining references to commons-logging, log4j, logkit, jboss.logging result from dependencies in hibernate, commons-beanutils, apache.servicemix

kasemir, 2016-04-06:
Created https://github.com/ControlSystemStudio/cs-studio/wiki/Logging to describe the suggested use.

willrogers, 2016-04-06:
There should be no further references to log4j or slf4j in cs-studio.

closed 2016-04-06 by willrogers