ControlSystemStudio/cs-studio #1674: 'unpack' plugins, for example jython

kasemir, 2016-03-18:
The feature.xml supports an unpack attribute `<plugin id=".." unpack="true">`. That plugin is then not a jar but a directory in the resulting product.
I've used that for at least two plugins:
* jython, because a considerable amount of Jython startup is in loading the standard *.py code, which takes longer when it's in a zip
* the product plugin, because that allows looking at the built-in plugin_customization.ini

Tycho ignores the unpack attribute in the feature.xml and instead uses `Eclipse-BundleShape: dir` in the manifest, see https://wiki.eclipse.org/Tycho/Packaging_Types#eclipse-feature.

I'll try that and submit a pull request for the maven-osgi-bundles/repository/pom.xml

berryma4, 2016-03-18:
We had Eclipse-BundleShape: dir  set for jython in the past.  But, @jbobnar had an issue, so it got switched back.  I believe it had to do with pydev.

berryma4, 2016-03-18:
I can't find the issue ticket about this change though.

berryma4, 2016-03-18:
added:
https://github.com/ControlSystemStudio/maven-osgi-bundles/commit/4b85167c32c1d082f01d891d64fde4f481251f42

removed:
https://github.com/ControlSystemStudio/maven-osgi-bundles/commit/c610de9dcf904eef191d1cdd2800195f3ff6c4a0#diff-2e643641016aae22c4e88bc4861b4bacL183

kasemir, 2016-03-18:
Interesting. I thought jython was expanded because I had put the 'unpack' into my features.
Today I spent time going through tycho docs to see why that's ignored and what to do about this, and it turns out you had already done that.

Anyway, I still have operational CSS setups on beamlines based on the older versions, where jython is expanded and Pydev is also installed. So I wonder what the problem is with that.

jbobnar, 2016-03-19:
One issue was the incredible size of unpacked jython bundle (if I remember correctly it was more than 120MB). The second issue was with pydev.

Pydev comes with its own jython library (org.python.pydev.jython bundle). As long as you are using only that one, its fine. I don't know the actual use/test -case (I was not involved in this), but they're doing something with jython and calling some scanning libraries. That doesn't work because scanning is using the org.python.jython from the maven-osgi-bundles. I don't remember if it was actually a linking constraint violation or it simply didn't work (whatever the reason). The solution that ITER adopted was to replace the pydev jython interpreter with the one from org.python.jython. Unfortunately, that only works if the bundle is packed into a jar (it's a pydev limitation).

kasemir, 2016-03-21:
The size of the  jython bundle is exactly the point: Un-zipping that each time a jython interpreter is started is a big part of the jython startup time. So having it already in expanded form results in faster startup of each *.opi that uses jython.

You're right that pydev includes its own org.python.pydev.jython bundle. In my operational setups, I find both the org.python.jython that we add to CSS and the org.python.pydev.jython  from pydev in expanded form. So far that has not caused problems. Maybe that's because they never really mix?
The scripts executed from *.opi files run under the org.python.jython bundle, they're started by the opibuilder code to get access to the widgets etc.
One could also 'run' python code under the org.python.pydev.jython from within pydev, but that doesn't really "work" for scripts because they don't have access to widgets and PVs.
Or have you found a way to start a script from the opibuilder, then set breakpoints etc. and debug it with pydev, and that way the two jython libraries get mixed?

jbobnar, 2016-03-22:
I backtraced the changes in ITER's repository and bugzilla. It appears that the jython plugin that comes from pydev does not work for them, because it is missing some modules (it's not a linking issue as I thought). The org.python.jython does the job fine and ITER is using that one. But it can only be used if it is packed in a jar. Unpacked jython.jar cannot be used by pydev. [This](https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/html/script/img/PyDevPreferences.png) is not valid if the jython is unpacked.

When tycho build was introduced the org.python.jython plugin was a jar file. A few months later it became unpacked and things stopped working for ITER. We asked Eric if the jar could be packed again. None of us was aware of any performance issues. I just did a quick test executing a script from BOY using a packed/unpacked org.python.jython jar. In the very first time that a jython script is loaded I didn't see any difference at all. In both cases it takes 3-4 seconds to start. In all subsequent cases, the unpacked version is slightly faster: 400 ms (unpacked) vs 700 ms (packed). If this is relevant, then by all means change the plugin to unpacked. We'll add another instance of the jython.jar so that it can be used by the interpreter.

closed 2016-03-23 by kasemir

kasemir, 2016-03-28:
Interestingly, http://bugs.jython.org/issue2355 describes how Jython can only use /Lib from an unpacked jar. So depending on where you look, you find claims that the /Lib for jython must be in a jar or must not be in a jar.