ControlSystemStudio/cs-studio #1675: thirdparty build fails because of olog/jersey

berryma4, 2016-03-18:
see:
https://github.com/ControlSystemStudio/cs-studio-thirdparty/issues/14

@shroffk, is this from your channelfinder changes?

kasemir, 2016-03-18:
I think it's because of the profile changes around "!cs-studio" or "cs-studio".

There is a way to specify a csstudio.local.repo, but not a csstudio.composite.repo.
We had some discussion about using csstudio.composite.repo and removing the csstudio.local.repo.

cs-studio-thirdparty/pom.xml now has the "!cs-studio" which causes it to skip downloads from http://download.controlsystemstudio.org/maven-osgi-bundles
--> Good, because I want to use my local ones

.. but it lacks the csstudio.composite.repo support
--> So I get the missing jersey

kasemir, 2016-03-18:
I'll try to add support for csstudio.composite.repo to cs-studio-thirdparty/pom.xml and submit a pull request.

shroffk, 2016-03-18:
Thanks a lot for this Kay, Sorry about the failing build...I think this comes back to the Issue https://github.com/ControlSystemStudio/cs-studio-thirdparty/issues/7
The maven remote repo was hard coded since I did not have a way to access it via composite repo.

Mow all the cs-studio main repos support compostite repos, can we mark the local.repo as deprecated in all the other poms too?

closed 2016-03-18 by berryma4

kasemir, 2016-03-18:
If we 'deprecate' it, that just means spending half an hour adding all the `<-- Deprecated 'cause, you know,  .. -->` comments, getting a pull request, merging it, .. and then it's ignored.

We might instead just _remove_ the local.repo support, with a commit message "use composite".

shroffk, 2016-03-18:
Ok,
Even better, I was just worried about breaking someone elses build system.
Do you want me to do this?

kasemir, 2016-03-18:
Please, if you have time!

Yes, there is a chance to break the build. But the discussion on https://github.com/ControlSystemStudio/cs-studio-thirdparty/issues/7 could serve as the "it's deprecated, we told you so".