ControlSystemStudio/cs-studio #1686: How to contribute page is hopelessly out of date

willrogers, 2016-03-21:
This page is incorrect.

https://github.com/ControlSystemStudio/cs-studio/wiki/HowToContribute

Does anyone mind if I update it and link from the front page of the wiki?

@kasemir @berryma4 @shroffk @jbobnar

kasemir, 2016-03-21:
It's actually amazingly up to date, all ideas still apply, just needs links to github instead of sourceforge ;-)
By all means, go ahead and update.

willrogers, 2016-03-22:
I have:

* Updated https://github.com/ControlSystemStudio/cs-studio/wiki/HowToContribute
* Added https://github.com/ControlSystemStudio/cs-studio/wiki/CodingGuidelines
* Added them to the front page of the wiki https://github.com/ControlSystemStudio/cs-studio/wiki

willrogers, 2016-03-22:
You can close this ticket if you like.

berryma4, 2016-03-25:
I added a fork part, and something about the meetings.
Also, I fixed the link "How to get Started" to point to this wiki page.
http://controlsystemstudio.org/contribute.html

Thank you @willrogers !

closed 2016-03-25 by berryma4