ControlSystemStudio/cs-studio #1689: Errors when modifying Channel Finder data with Channel Viewer

dylan171, 2016-03-23:
@shroffk 

While testing the Channel Finder v3 integration with CS Studio, I have been able to view/search the channel data ok, but modifying the channel property/tags causes errors:

Modify property:
`2016-03-23 10:52:25.936 INFO [Thread 46] org.csstudio.utility.channel.actions.ModifyChannelJob (run) - modified propertygov.bnl.channelfinder.api.Property@b20410bd
com.sun.jersey.api.client.UniformInterfaceException: PUT https://localhost:8181/ChannelFinder/resources/properties/pvStatus returned a response status of 500 Internal Server Error
	at com.sun.jersey.api.client.WebResource.voidHandle(WebResource.java:709)
	at com.sun.jersey.api.client.WebResource.access$400(WebResource.java:74)
	at com.sun.jersey.api.client.WebResource$Builder.put(WebResource.java:524)
	at gov.bnl.channelfinder.api.ChannelFinderClientImpl$UpdateChannelProperty.run(ChannelFinderClientImpl.java:854)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)`

Add Property:
`com.sun.jersey.api.client.UniformInterfaceException: POST https://localhost:8181/ChannelFinder/resources/properties/TestProp returned a response status of 500 Internal Server Error
	at com.sun.jersey.api.client.WebResource.voidHandle(WebResource.java:709)
	at com.sun.jersey.api.client.WebResource.access$400(WebResource.java:74)
	at com.sun.jersey.api.client.WebResource$Builder.post(WebResource.java:555)
	at gov.bnl.channelfinder.api.ChannelFinderClientImpl$UpdateProperty.run(ChannelFinderClientImpl.java:935)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)`

Add Tag:
`com.sun.jersey.api.client.UniformInterfaceException: PUT https://localhost:8181/ChannelFinder/resources/tags/TestTag returned a response status of 500 Internal Server Error
	at com.sun.jersey.api.client.WebResource.voidHandle(WebResource.java:709)
	at com.sun.jersey.api.client.WebResource.access$400(WebResource.java:74)
	at com.sun.jersey.api.client.WebResource$Builder.put(WebResource.java:524)
	at gov.bnl.channelfinder.api.ChannelFinderClientImpl$UpdateTag.run(ChannelFinderClientImpl.java:801)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)`

shroffk, 2016-03-23:
I think these methods need scripts support enabled on elasticsearch

I have updated the installation instructions on how to do this
https://github.com/ChannelFinder/ChannelFinderService/blob/master/README.md

Thanks

dylan171, 2016-03-24:
I'm probably the worst person to test the Channel Finder /CS Studio integration because I haven't used it that much. In the past we never used the CS Studio interface for modifying Channel Finder data.

Enabling scripting on ElasticSearch helped.

I can successfully add a tag to channel, but when I add the same tag to another channel, it disappears from the first channel. 

Attempting add a property results in an exception:
`com.sun.jersey.api.client.UniformInterfaceException: POST https://localhost:8181/ChannelFinder/resources/properties/myStatus returned a response status of 400 Bad Request
	at com.sun.jersey.api.client.WebResource.voidHandle(WebResource.java:709)
	at com.sun.jersey.api.client.WebResource.access$400(WebResource.java:74)
	at com.sun.jersey.api.client.WebResource$Builder.post(WebResource.java:555)
	at gov.bnl.channelfinder.api.ChannelFinderClientImpl$UpdateProperty.run(ChannelFinderClientImpl.java:935)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)`

Modifying an existing property does not throw an exception and it updates the property on the correct channel, but it removes/clears the property on all other channels.

It might be important to know that I'm using ElasticSearch v1.7.5.

dylan171, 2016-03-24:
Let me know if there is anything I can do to help with the debugging.

shroffk, 2016-03-24:
1. The tag issues was related to this
https://github.com/ChannelFinder/javaCFClient/issues/5

I have submitted the pull request
https://github.com/ControlSystemStudio/cs-studio-thirdparty/pull/18

2. The properties should be working fine. I currently have a test setup where I am using the addproperty command to update a property value on a set of channels.

are you using the addproperty or the edit channel?

dylan171, 2016-03-24:
When using addproperty everything seems to work as expected.
(I couldn't reproduce the previous exception)

But when using modify channel, it sets the property value correctly for selected channel(s), but then clears/removes the property for all other channels.

shroffk, 2016-03-24:
Ok, 
at least we have consistency...I will look into the edit channel command and ensure that it is following the changes in the service API now