ControlSystemStudio/cs-studio #1702: Integrate Dawn CPython interpreter in CS-Studio

willrogers, 2016-04-01:
The principal use of this would be to execute dynamically certain scripts and possibly return data.  Dynamic plotting of data using a Python interpreter would also be useful.  CPython is important because Numpy specifically is a usfeful tool.

Dawn (http://www.dawnsci.org/) does these things and we could try to integrate the relevant features.

EASE is an Eclipse project that does a similar thing with Jython and Javascript.  [This article](https://opensource.com/life/16/2/how-use-python-hack-your-ide) is a good introduction.  It may be that someone will try to integrate CPython into EASE, but this doesn't appear to be happening right now.

Related to this is Dawn's datasets, which are becoming an independent Eclipse project, known as January: https://projects.eclipse.org/proposals/january.

kasemir, 2016-04-01:
In my understanding, the CPython integration into EASE is planned after the data types used for the communication between CPython and the Java proxi have been settled, i.e. the January project.

willrogers, 2016-05-31:
I had a meeting with some of the Dawn developers.

* They think it should be straightforward for them to provide a minimal feature including the Dawn CPython interpreter, that we could include
** Out of the box, this would just be a console that we could use for plotting - i.e. not really any integration with CS-Studio.  How else would we like to be able to integrate?
* @gerring said that EASE was complementary to the Dawn Python integration, rather than the same thing.  EASE allows you to script some of the insides of Eclipse.  Is this closer to what we want?

What I always have difficulty with is understanding _how we get hold of_ any data we want to use in Python, and then how we send it back to Java, if that's what we want to do.  In Dawn, you can use the names of the plots to retrieve data or get it directly from files.

willrogers, 2016-06-01:
Conclusion:  a Python interpreter from which you can plot is a useful first step.  After we have that, we could investigate integration, particularly:

* Data from the archiver
* Scan server

berryma4, 2016-10-05:
Closing, because CPython support was added to the Display Builder, in which all current use cases involve.

closed 2016-10-05 by berryma4