ControlSystemStudio/cs-studio #1704: Release 4.3.x (master)

berryma4, 2016-04-04:
Proposing to freeze the master branch to 4.3.x.

 - [ ] BOY, mostly: Rethink Perspectives/Compact Mode/…, use Views for runtime #544 
 - [ ] Save and share perspectives 
 - [ ] Databrowser Archive refresh issue
 - [ ] running 4.3 with a 4.1 workspace
 - [ ] Diirt time
 - [ ] remove desy code
 - [ ] remove products
 - [ ] remove slf4j
 - [ ] getting a view in a specific perspective ... example opening an opi from the navigator not in the same spot in as in the previous versions

Site Representatives, are there any other outstanding issues that need to be addressed before release of 4.3.x?
@berryma4, @kasemir, @shroffk , @utzeln, @mjclarke01, @charlesHenryPATARD, @ksdj, @xinyuwu, @jpenning, @willrogers

berryma4, 2016-06-15:
@shroffk is working on "Databrowser Archive refresh issue", although this needs an issue ticket.

berryma4, 2016-06-15:
I'm working on testing #1698 today, which will close the macro item.

berryma4, 2016-06-16:
@shroffk 
Regression tests, I see a couple of possible issues:
Switch rule: 
   3rd down, doesn't accept new values
   5th down, value needs to be reentered if written while switch is down
   6th down, value needs to be reentered if written while switch is down

ComboBox test:
   Non-editable combo box, is it suppose to be editable for 1-3?

file-python:
   cvs files missing
   when trying to edit/save with new cvs file
   Save Failed
      org/diirt/util/stats/Range

berryma4, 2016-06-16:
@willrogers I know we have discussed this before, but is there anything we can do about the blank editor window popping up every time an opi is opened?
People are going to go crazy about this.

willrogers, 2016-06-16:
We've talked about this a couple of times.  The conclusions we came to:

* if you open a file _from the navigator_ the editor pane is restored (this happens for e.g. both standalone window and workbench)
* most user workflows __don't__ use the navigator
* @kasemir looked into this and the open call is hard-coded in the Eclipse source
 * @kasemir idly suggested we could try using Science WG contacts to get them to consider a change to Eclipse

It's explicitly bad for the user experience (it's unexpected and useless and obvious) but the latest conclusion was that it didn't affect users often and it wasn't possible to fix.

kasemir, 2016-06-16:
That's #1410 . We know that the problem is caused by org.eclipse.ui.internal.WorkbenchPage calling "setEditorAreaVisible(true)" even for 'external' editors.

I have no idea how to change that.

Only option would be to work around it in our code to
1) Check if the editors area was visible
2) Then open the view
3) Hide the editors area if it wasn't visible at step 1

berryma4, 2016-06-16:
@shroffk 
Regression tests update
ControlSystemStudio/opi-regression-tests@ebd19eed815f22eff26f1d83a8c1978768881f1f
fixed tests
Switch rule: 
3rd down, this only accepts strings, added description
5th down, this didn't correct the precision if entered when off, then turn on, it needed a trigger on added
6th down, same as 5th down

ComboBox test:
Non-editable combo box, is it suppose to be editable for 1-3?  No.
It is not editable ever.

So the only really error is:
file-python:
cvs files missing
when trying to edit/save the opi file
Save Failed
org/diirt/util/stats/Range

berryma4, 2016-06-16:
@kasemir 
Also, I can't save a databrowser plot? (or save as)

berryma4, 2016-06-22:
I've submitted #1813 PR to close the orbit punch list item.

berryma4, 2016-06-30:
https://docs.google.com/forms/d/1cKILK51oWpVcy5iDCe4gPK6mvCSvlMWIS5JazuO2_Lc/viewanalytics?usp=form_confirm

| Test       | Result           |
| ------------- |:-------------:|
| timestamp | 12/22/2015 12:43:42 |
| issue ticket | https://github.com/ControlSystemStudio/cs-studio/issues/1540 |
| Tester | berryman |
| __BOY Examples__ |
| All BOY Examples | Pass |
| __Issues__ |
| Issue 1081 | Pass |
| Issue 1103 | Pass |
| Issue 1197 | Pass |
| Issue 1656 | Pass |
| __Performance Tests__ |
| graph_performance_test.opi  | Pass |
| large_led_performace_test.opi | Pass |
| small_led_performace_test.opi | Pass |
| tank_performace_test.opi | Pass |
| test_update_performance_test.opi.opi | Pass |
| __Rules Tests__ |
| General_rule.opi | Pass |
| Graphs_rule.opi | Pass |
| LED_color_rule.opi | Pass |
| Label_rule.opi | Pass |
| Switch_rule.opi | Pass |
| __WidgetTests__ |
| ActionButton_test.opi | Pass |
| BubbleGraph_test.opi | Pass |
| ComboBox_test.opi | Pass |
| Container_test.opi | Pass |
| Gauge_Meter_Thermometer_Tank_test.opi | Pass |
| LED_test.opi | Pass |
| TextInputWidget_test.opi | Pass |
| TextUpdate_test.opi | Pass |
| XYGraph_test.opi | Pass |

berryma4, 2016-07-21:
@shroffk can you verify "running 4.3 with a 4.1 workspace", so I can close this.
I will make an issue for "getting a view in a specific perspective ... example opening an opi from the navigator not in the same spot in as in the previous versions"
to be addressed in a future version.

berryma4, 2016-07-25:
@shroffk, I clicking "running 4.3 with a 4.1 workspace" as okay.  I haven't seen an issue.  If you have one, please open a new ticket.

berryma4, 2016-07-25:
closing/checking "getting a view in a specific perspective ... example opening an opi from the navigator not in the same spot in as in the previous versions"
#1832

berryma4, 2016-08-03:
I had issues with Jenkins, so TravisCI deploys instead.
#1843

closed 2016-08-03 by berryma4