ControlSystemStudio/cs-studio #1735: DataBrowser add hot keys for zoom in/out

berryma4, 2016-04-11:
Suggestion of improvement from Survey
Survey entry 39:
An improvement to data broser:  It's a pain always to have to select the tool, THEN do the action. A better setup would be something like: "Shift+mouse = zoom in; alt+mouse=zoom out; mouse=move around."

kasemir, 2016-04-11:
While I understand the motivation for this, we also need to avoid unintended side effects where novice users are surprised when the plot suddenly changes on them.

In the plain "pointer" mode, i.e. the one where no zoom option is selected, simply clicking or moving the mouse should not suddenly result in a pan or zoom. It needs to remain the mode where you can move the mouse with the cross-hair cursor to see values, or to grab & move annotations without otherwise affecting the plot.

Using ctrl, alt or shift, however, could allow pan/zoom.
Should be somewhat easy to remember. Not sure if there is already a good commonly known example.

How about:
Pointer mode + no modifiers: As before, no pan/zoom.
Shift + mouse move: pan, because it _shifts_ the plot around.
Control + mouse start..end drag: Zoom in. You _control_ what you see in more detail.
Alt + mouse click: Zoom out. Using Alt because it's the only modifier left.

nickbattam, 2016-04-11:
@kasemir Could we use a mouse+keyboard pan/zoom controls users are already likely to be familiar with?

e.g. Microsoft Visio:
    Hold down both the CTRL and SHIFT keys. The pointer turns into a magnifying glass.
        Click the left mouse button to enlarge the view.
        Click the right mouse button to reduce the view.
        Click the left mouse button and drag a rectange that contains just the region you want to zoom into.
        Right-click and drag to pan the diagram.

willrogers, 2016-04-11:
Surely control and mouse-wheel is standard for zooming.

kasemir, 2016-04-14:
I started to work on this based on the Visio ideas for the new JavaFX-based plot, but now reconsider.  

Not sure if the original requester actually understood that you can zoom/pan the whole plot, the X axis, or a selected Y axis. That's why you first select the tool (like zoom "in"), and then click either the plot or an axis to zoom that aspect of the display. Archive access, even when "fast", is still somewhat slow. Time saved by quickly click-click-click zooming is lost when you're waiting for the data to be updated, so one extra click and then carefully adjusting the zoom is usually better than rapidly zooming around.

Using Ctrl-'+' and Ctrl-'-' keys do seem universal to zoom in/out, they could be mapped to the main plot area. I also agree with the mouse wheel, again mapped to the main plot area.

For combinations of ctrl/shift/alt and mouse clicks I see no standard. Visio, Tinkercad, Autocad, .. are all different. Visio isn't more of a standard than the rest. I've never used it, and wasn't mentioned by the original requester, either. Using the right mouse button to zoom out can conflict with context menu activation.

So now I'm leaning towards:

Ctrl-'+' and Ctrl-'-' to zoom plot area in/out.
Mouse wheel to zoom plot area in/out.

-> Standard, nobody will argue about that.

Holding Control or Alt basically holds the "Zoom in" resp. "Zoom Out" button pressed down.

-> Not a standard, but then there isn't one. If you can't remember and just try pressing Ctrl, shift, alt, .. you'll see how the tool bar buttons activate, and thus know what will happen when you then click or drag the mouse in the plot or axes areas.

willrogers, 2016-04-15:
I agree with the above.  It would also be nice that if you mouse-wheeled when you were over the an axis you could zoom only that axis.  Zooming only one axis is a feature that our users don't find out by themselves but really like once they know.

Just to hijack the ticket, would it be possible to change the mouse cursor depending on what function is selected?  The existing cursors don't seem to match the setting except for the standard pointer one.

kasemir, 2016-04-15:
Has been implemented in the display builder update (#1230).
Zooming w/ mouse wheel acts on axis or plot, whatever's below the mouse pointer at the time.
Cursor changes as well.
Additional keyboard shortcuts to open/close toolbar etc.

In principle, it shouldn't be too hard to duplicate that in the SWT version of RTPlot.

[image]

berryma4, 2018-04-06:
This feature is in databrowser 3.

closed 2018-04-06 by berryma4