ControlSystemStudio/cs-studio #1738: Unfavorable survey response

berryma4, 2016-04-11:
Survey line 37:
Only the look of it, the feel of it, all interactions with it, the speed (or lack of) it.  Other than that it's amazing.

berryma4, 2016-04-11:
Maybe we could say the new Display Builder is a solution.

kasemir, 2016-04-14:
User should write his/her own user interface in Qt, because that will by definition result in "a user interface that is beautiful on every screen and performs perfectly on every platform" (http://www.qt.io).

shroffk, 2016-04-14:
Could we close this ticket since it really isn't a very actionable item...

Usually, with such a unfavorable experience could be address by have more/better training about how one should and should not use cs-studio...so that could be an Issue ticket.

The above feedback seems to indicate an incorrect choice to tool/workflow. Maybe a webclient with js  or a QT client like Kay recommended might be a better fit.

CS-Studio priorities - talk to multiple control systems, multiple epics and high level services, provide integration between these applications.

berryma4, 2016-04-14:
I think we've addressed this the best we can.  All the above are good points.  Maybe CS-Studio is not being used "correctly", or what the user wants to do is not in our scope/objective.  So, it seems the answer is to improve training and make our scope more clear.  Which @shroffk has added the new training guides, and we have tightened our scope in the last management review.

berryma4, 2016-04-14:
If we have similar complaints in the future, we can review the above for effectiveness.

closed 2016-04-14 by berryma4