ControlSystemStudio/cs-studio #1739: Unfavorable survey response

berryma4, 2016-04-11:
Survey line 40:

Complete lack of GUI standards makes the screens hard to understand.  A control system needs to be presented to the users in a consistent fashion, not a a mishmash of ideas from a variety of people that have little concept in building GUIs.  Very few applications are available.  The user interface is confusing.  Screens are difficult and time consuming to build. Screens require IOC programming because using the logic features overwhelm and kill the program.  Screen version management control is poor.  Software is buggy and crashes often.  The CS-Studio interface is usually larger than the screens being displayed, which wastes monitor space for other applications.  Charts default to "Area" plots which are rarely used, instead of line plots which are what people want.  Clicking on a control line for a chart item causes the associated color to be obscured.  Lots of little annoying things like those above make the software frustrating for users.

berryma4, 2016-04-11:
I numbered the items to make addressing easier.  Although, in general, most of this sounds like a training issue.

1. might be a lab organizing issue?  But, the widget "classes" might help.

2. There is a save/restore and tuning application in 4.3.  I'm not sure we are in the business of scaling.  And, we do have BEAST as an alarm handler.

kasemir, 2016-04-11:
1. You're right. Unfortunately not everybody agrees with me on how things should look.

6. Need specific bug/crash report. We use CSS in operation and obviously depend on it running reliably. For us, it does. If it crashes at your place, you need to provide more detail. And you are of course more than welcome to join the development and help us fix such crashes.

8. Don't mistake your opinion with "what people want". Besides, default data browser trace type is configurable. Still, if you read the online help on optimized data retrieval and learn more about the underlying archive system, you may reconsider what you want as a default.

9. Was fixed with #774, ~Dec. 2014

closed 2016-06-21 by berryma4