ControlSystemStudio/cs-studio #1741: Add new "Project" workflow

berryma4, 2016-04-11:


berryma4, 2016-04-12:
Site Representatives, please okay the new "Project" (Design and Development) workflow, or comment to edit.
 - [x] @berryma4 
 - [ ] @kasemir 
 - [ ] @shroffk  
 - [ ] @utzeln 
 - [ ] @mjclarke01 
 - [ ] @charlesHenryPATARD 
 - [ ] @ksdj 
 - [ ] @xinyuwu 
 - [ ] @jpenning 
 - [ ] @willrogers

mattclarke, 2016-04-12:
Would it be better if it said the branch name **started** with "issue_xx" rather than **contains**?
Other than that it looks good to me

charlesHenryPATARD, 2016-04-12:
- [x] @charlesHenryPATARD

Ok for me.

berryma4, 2016-04-12:
@mattclarke, I can make that change.

berryma4, 2016-04-12:
[Design and Delevopment.pdf](https://github.com/ControlSystemStudio/cs-studio/files/215667/Design.and.Delevopment.pdf)

berryma4, 2016-09-08:
We have majority

closed 2016-09-08 by berryma4