ControlSystemStudio/cs-studio #1742: Modified Quality Policy and Objectives

berryma4, 2016-04-11:
Was:

>  Policy:
> Through a strategy of continuous improvement and teamwork, the Control System Studio Collaboration is dedicated to supplying tools for data integration and data visualization to enable our users to achieve their scientific objectives.
> 
> The foundation for achieving our commitment is based on:
> 
>  - understanding and meeting the requirements of our users,
>  - continuously improving all processes related to the Control System Studio product,
>  - effectively utilizing the creative talents in the collaboration,
>  - and meeting statutory, regulatory, and other requirements.
> 
> Objectives:
> Fraction of releases per site deployed vs releases considered for deployment. Min 75%
> 100% sites with registered site representatives moved to Tycho
> 

Proposed:

> Policy:
> Through a strategy of continuous improvement and teamwork, the Control System Studio Collaboration is dedicated to supplying control system tools for machine and experiment operator interfaces, automation and service integration to enable our users to achieve their scientific objectives.
>
> The foundation for achieving our commitment is based on:
>
>  - understanding and meeting the requirements of our users,
>  - continuously improving all processes related to the Control System Studio product,
>  - effectively utilizing the creative talents in the collaboration,
>  - and meeting statutory, regulatory, and other requirements.
> 
> Objective:
> 90% User satisfaction from registered sites with a site representative
> 90% Developer satisfaction from registered sites with a site representative

berryma4, 2016-04-12:
Site Representatives, please okay the proposed policy/objective changes, or comment to edit.
 - [ ] @berryma4 
 - [ ] @kasemir 
 - [ ] @shroffk  
 - [ ] @utzeln 
 - [ ] @mjclarke01 
 - [ ] @charlesHenryPATARD 
 - [ ] @ksdj 
 - [ ] @xinyuwu 
 - [ ] @jpenning 
 - [ ] @willrogers

mattclarke, 2016-04-13:
I am happy with this

berryma4, 2016-04-13:
@mattclarke , should I be using @mjclarke01 or @mattclarke ?

mattclarke, 2016-04-13:
@mattclarke is my work account, so probably best to use that please.

berryma4, 2016-09-08:
We have a majority

closed 2016-09-08 by berryma4