ControlSystemStudio/cs-studio #1744: uploadRepo doesn't make sense for core and applications

willrogers, 2016-04-12:
If you specify `-PuploadRepo` on a cs-studio build the repos will be uploaded to a particular location.

This doesn't make sense for `core/p2repo` and `applications/p2repo` because the `compositeContent.xml` refers to locations which do not exist when the other repos have been uploaded to that same location e.g.:

* base repo is uploaded to `<upload.root>/core/4.3/base`
* `compositeContent.xml` contains `../base/base-repository/target/repository`

This doesn't actually break anything right now, just those composite repos are there and don't make sense.  It would be nice if we could upload a composite repo that did make sense for core and applications.

berryma4, 2016-04-12:
Yes, p2repo seems to be set to the local build instead of the upload.
Should we make two composites (one for the upload uri, one for the local target), or should we force people to use a file path (<upload.root>/core/4.3/base) for local development.

willrogers, 2016-04-12:
Ideally you could use the repo locally but `-PuploadRepo` would send the correct files for a remote repo.  For this you'd need two different `composite*.xml` files though and I'm not sure how you'd do that.

berryma4, 2016-04-13:
#1726

berryma4, 2016-04-15:
Okay, I changed the p2repo directory's pom.  As you said, it doesn't make sense to upload the development composite paths.  So I made new composite*.xml files, and upload those.

closed 2016-04-15 by berryma4