ControlSystemStudio/cs-studio #1745: Management Review Report

berryma4, 2016-04-12:
Here is the Management Review Report from the review we had during the code a thon:
https://docs.google.com/document/d/1FhKhnjsFPFSBCrztP4WpXJly4dfLav4-y8Rz_DcTJWY/edit

Please comment on any additions, or check to call it done.

 - [x] @berryma4 
 - [ ] @kasemir 
 - [ ] @shroffk  
 - [ ] @utzeln 
 - [ ] @mjclarke01 
 - [ ] @charlesHenryPATARD 
 - [ ] @ksdj 
 - [ ] @xinyuwu 
 - [ ] @jpenning 
 - [ ] @willrogers

Thank you everyone for your time!

willrogers, 2016-04-12:
The bit on pages 4-6 doesn't seem to have been updated.

berryma4, 2016-04-12:
Thank you, I removed the older sections.

jpenning, 2016-04-12:
Thanks, Eric - no further additions from my side ... as you might have 
suspected :-)

Jörg.

kasemir, 2016-04-12:
Updated final section about idea/motivation for turning display builder into Eclipse project.

berryma4, 2016-09-08:
We have a majority

closed 2016-09-08 by berryma4