ControlSystemStudio/cs-studio #1746: centralrepository is incomplete in maven-osgi-bundles

willrogers, 2016-04-12:
I'm not sure why, but:

In maven osgi bundles, the centralrepository p2 repo only contains `org.csstudio.platform.libs.epics_4.3.0.20160412145532.jar`.

For some reason, the other jars from the repository p2 repo are not mirrored across.

This means that `-PuploadRepo` does not work correctly.

p.s. Should this ticket be here or there?

berryma4, 2016-04-12:
centralrepository is a diff of repository and your uploaded target with an appended index with the new binaries.  It would make sense that org.csstudio.platform.libs.epics would be the only one, because it has a snapshot version and always has a different version on build.   You probably need to change your "central url" property to pint to your repo.

willrogers, 2016-04-12:
Right, I'm sure you told me this last week but I didn't understand.

If I understand now, this means that we should use the same property as for the `upload.root` for the `cs-studio.central.url` property, BUT the protocol is different ...

berryma4, 2016-04-13:
#1726

closed 2016-04-13 by berryma4