ControlSystemStudio/cs-studio #1750: LogEntry search and java 8 time format

shroffk, 2016-04-13:
The old logbook api (o.c.logbook/LogbookClient) allowed you to define search string with the following format
*some text* logbook:operations tags: abc, efg from:last 8 weeks
but this does not work very well with time defined as "2016/04/12 14:11:00.000" since it consists ":" 

I want to suggest a change in the search API to switch the use of ":" by "="

is this change acceptable
 - [ ] NSLS2 (olog) 
 - [ ] SNS
 - [ ] Diamond

@berryma4: could you assign the appropriate reviewers

kasemir, 2016-04-14:
"14:11" is a very natural way to define a time, would be a real pity to change that just for the logbook search.

Isn't is safe to assume that the search key words will always be text, not numbers?

Using a regular expression like `[a-z]+:` would thus detect search key words while still allowing "14:11" as a value: `from: 14:11`.

shroffk, 2016-04-14:
Just to avoid confusion..the change I was requesting would not change the way time is represented.
I want that to be as consistent as possible across cs-studio.

I was recommending changing
*some text* logbook=Operations tags=Fault from=-8 hours to=now

shroffk, 2016-04-14:
Reverting back to the old representation,

`search for some text logbook:Operations, LOTO,Commissioning tags: testTag, testTag2 from:2016/04/12 14:11:00.000 to: 2016/04/12 15:11`

Using Kay's suggestions changing regular expression to not allow keyword with digits. This seems like a less painful change than previous one.

berryma4, 2016-04-14:
I like the colons better.  It looks nicer, and it's what google uses (so "people" might be more familiar with it).

closed 2016-04-25 by shroffk

willrogers, 2016-04-26:
Sorry to be late this sounds fine to me.