ControlSystemStudio/cs-studio #1751: Databrowser: delete trace that has been hidden

berryma4, 2016-04-13:
If you try to delete a trace that was unchecked "show", it doesn't remove itself from the list.  If you show/hide another trace in the list, the table updates with the trace removed.

kasemir, 2016-04-13:
I've tried to reproduce like this:

 * Create new data browser plot
 * Via "Add PV" context menu, add "sim://sine"
 * Add "sim://ramp" (both PVs added with default settings, so they're on their own axis)
 * Press the "Stagger" button once so both traces are nicely shown
 * In Properties view, un-check the "Show" box on the sim://ramp PV, say "Yes" to the warning dialog. (trace is now hidden)
 * In Properties view, use context mennu on the not-shown sim://ramp trace, select "Delete"

-> For me, the trace is removed from the list.
I can then also use "Undo" to add it back in, another "Undo" to show it again, then "Redo" to hide and finally delete is again. Tried with latest sources on Mac and Linux.

kasemir, 2016-04-14:
If you can still reproduce, is there by chance a message in the log that would shed a light?
Like in #560?

berryma4, 2016-04-14:
The problem is in 4.1.x windows, and I can not reproduce it in 4.3.x (current master).  Closing ticket.

closed 2016-04-14 by berryma4