ControlSystemStudio/cs-studio #1755: Mockito from Orbit is problematic because it is signed

jbobnar, 2016-04-14:
Commit https://github.com/ControlSystemStudio/maven-osgi-bundles/commit/a969c2be4bf919e94a8def8ff3e77b5a8a5d7c60 suggests that mockito from orbit should be used. That bundle is problematic, because it is signed and the result is that junit tests can no longer be run as simple junit tests in eclipse. Instead they need to be executed as "plugin junit tests" or tycho has to execute them. That takes much longer to start than simply executing the code as a junit test.

http://stackoverflow.com/questions/28579999/securityexception-when-running-plain-junit-mockito-in-eclipse-rcp-project

berryma4, 2016-04-14:
It looks like the fix was released Oct 2015:
https://github.com/cglib/cglib/commit/bb211dd47b557d76ccd477e403d703c82bac2b04
http://search.maven.org/#artifactdetails%7Ccglib%7Ccglib%7C3.2.0%7Cjar

But, it shouldn't matter:
https://github.com/mockito/mockito/blob/c4dbcc2a0eba80847caf70e5efbe87b5118af813/doc/release-notes/official.md#2014-beta-2015-06-12-2018-utc
since Mockito 2.0.14, they have removed cglib as the default.

And, since 2.0.32, they have made it more like they will not be supporting it as an alternative:
https://github.com/mockito/mockito/blob/c4dbcc2a0eba80847caf70e5efbe87b5118af813/doc/release-notes/official.md#2032-beta-2015-12-17-0750-utc

So, can we move up a version?

jbobnar, 2016-04-14:
Mockito 2.x is not yet on orbit, so you'll need to add it to the maven-osgi-bundles. Other than that, I think you can move up.

willrogers, 2016-04-15:
I had exactly this problem.  If a newer version of Mockito fixes this I would be very happy.

berryma4, 2016-04-15:
I doesn't increase in Neon, and I see a couple of open tickets in the Eclipse Bugzilla.  Maybe we need to add it back.  2.x is "beta" too.  So, 1.9.5 in maven-osgi-bundles?

jbobnar, 2016-04-18:
>  So, 1.9.5 in maven-osgi-bundles?

That works for me.

berryma4, 2016-07-21:
Sorry this took so long.

closed 2016-07-21 by berryma4