ControlSystemStudio/cs-studio #1760: Failed test: org.csstudio.archive.vtype.TimestampHelperTest.testRoundUp

berryma4, 2016-04-18:
Failed Jenkins test:

Error Message

Unable to obtain ZonedDateTime from TemporalAccessor: {},ISO resolved to 2012-01-19T12:23:14 of type java.time.format.Parsed

Stacktrace

java.time.DateTimeException: Unable to obtain ZonedDateTime from TemporalAccessor: {},ISO resolved to 2012-01-19T12:23:14 of type java.time.format.Parsed
	at java.time.ZoneId.from(ZoneId.java:466)
	at java.time.ZonedDateTime.from(ZonedDateTime.java:553)
	at org.csstudio.archive.vtype.TimestampHelperTest.testRoundUp(TimestampHelperTest.java:31)

shroffk, 2016-04-22:
I have seen this in simplepv, pvmanager probe and other places.

`    private DateTimeFormatter timeFormat = new DateTimeFormatterBuilder()
            .appendPattern("yyyy/MM/dd HH:mm:ss.n z")
            .toFormatter().withZone(ZoneId.systemDefault());`

Is this an acceptable solution? Should we have a apputil plugin which provides default formatters

kasemir, 2016-04-22:
That's a good solution!
I thought the fix was to first convert the Instant into a LocalDateTime, and then feed that to the formatter.
But if the 'zoned' formatter can directly handle the Instant, that's better.

Can such a formatter be in core/org.csstudio.java?

shroffk, 2016-04-26:
I know I am late but +1 to having these default formatter in o.c.java

closed 2016-04-27 by berryma4