ControlSystemStudio/cs-studio #1770: AnnunciationFormatter fails with the new format rules

shroffk, 2016-04-26:
At NSLS2 we use `{}` in our PV names, the new forced format option in cause parsing errors.

e.g. the Alarm description includes the pv name 
XF:31IDA-OP{Tbl-Ax:X1}Mtr

When this description goes through the new AnnunciationFormatter with the forced formatting, it causes a flurry of illegal argument exceptions since the formatter is trying to resolve the Tbl-Ax:X2 as a special custom format parameter in XF:31IDA-OP**{Tbl-Ax:X1}**Mtr

`
java.lang.IllegalArgumentException
	at org.eclipse.osgi.util.NLS.internalBind(NLS.java:192)
Caused by: java.lang.NumberFormatException: For input string: "Tbl-Ax:X2"
`

This was added due to 
Bugzilla ticket 7491: Alarm Table - alarm description without alarm severity prefix;

Potential solutions:
Could it be possible to make the forced formatting a preference?
or
Sanitize the format string before giving it to the NLS.bind

@kasemir @utzeln

kasemir, 2016-04-26:
Do you have a link to the commit or pull request for the new AnnunciationFormatter?

jbobnar, 2016-04-26:
@kasemir 
https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast/src/org/csstudio/alarm/beast/AnnunciationFormatter.java

@shroffk 
The commit you mention was related to this #1281. I'm sure this could be made a preference. I'll make a pull request with the change.

kasemir, 2016-04-26:
Bummer.

The use of '{0}', '{1}', .. in the annunciation strings has been in use for a while.
PVManager decided to use '{ .. }' for ill-advised modifiers like 'longString'; still, it's using curly brackets as well. Putting such brackets into the PV name was an unfortunate idea.

Still, too late now.

How about replacing the NLS.bind() calls with more specific String.replace() calls that replace only "{0}", "{1}", "{2}" with the respective replacement text?

shroffk, 2016-04-26:
> How about replacing the NLS.bind() calls with more specific String.replace() calls that replace only "{0}", "{1}", "{2}" with the respective replacement text?

I think this would be a good idea, to sanitize the input string before giving it to bind

```
format = format.replace("{", "'{'"); 
format = format.replace("}", "'}'");  
format = format.replaceAll("'\\{'([\\d]+)'\\}'", "{$1}");
```

This works on the following inputs

```
		System.out.println(process("aaa"));
		System.out.println(process("a23"));
		System.out.println(process("12312"));
		System.out.println(process("aaa{1}"));
		System.out.println(process("aa{1}a:a{2}"));
		System.out.println(process("a{s2}aa"));
		System.out.println(process("aa{sss{s}}a"));
		System.out.println(process("aa{sss{0}}a"));

OUTPUT:
aaa
a23
12312
aaa{1}
aa{1}a:a{2}
a'{'s2'}'aa
aa'{'sss'{'s'}''}'a
aa'{'sss{0}'}'a
```

This should not need a preference while still continue to support my unfortunate situation and NLS.bind()

shroffk, 2016-04-26:
This worked for me, @jbobnar can you double check and then we can make a pull request

kasemir, 2016-04-26:
How about much simpler?
```
message = format.replace("{0}", severity).replace("{1}", value);
```

shroffk, 2016-04-26:
Sure, If giving up the NLS.bind is fine with everyone.

kasemir, 2016-04-26:
The NLS.bind() was easy at the time. I don't have to keep it if the workaround is larger than the plain replace("{0}"). But just noticed that you already have a commit, so if that works, also fine.

jbobnar, 2016-04-27:
Both solutions are fine. However, if you choose replaceAll, I do recommend one change:
```
private static final Pattern PATTERN = Pattern.compile("'\\{'([\\d]+)'\\}'");

public static void format(String orgString, ...) {
   ...
   String newString = PATTERN.matcher(orgString).replaceAll("{$1}");
   ... 
}
```
You do not need to compile the regEx every time (it is expensive).

shroffk, 2016-04-27:
@jbobnar you are right I made the changes...
@kasemir Is there a reason the o.c.a.beast.tests fragment is removed from the plugin list?

kasemir, 2016-04-27:
You're right for pre-compiling the `replaceAll` pattern.

I had always thought that the `replace` call which only allows plain strings, no regular expressions, is faster. But at least for my copy of Java 8 it also ends up calling `Pattern.compile()...` with a `Pattern.LITERAL` flag. Not sure if that flag makes it a lot faster, or if pre-compiling is still helpful.

kasemir, 2016-04-27:
Looks like the o.c.a.beast.tests has been excluded for a while.
It certainly has "HeadlessTests" that would have to be renamed "IT", and the parent pom link also changed.
I'll update it.

shroffk, 2016-04-27:
I am already doing that...vcause I wanted to add testes for the formatter

kasemir, 2016-04-27:
Well, I also did it, then read your reply.

shroffk, 2016-04-27:
Oh...should we let Eric do the merge now

kasemir, 2016-04-27:
I'll delete my pull request.
You have the added test.
Only other difference: You @Ignore one test that I changed into IT.
So I'm good with your changes.

closed 2016-04-28 by shroffk