ControlSystemStudio/cs-studio #1788: scan server large scan returns 500

berryma4, 2016-05-12:
I get the following error with a large scan

``` java
ERROR:  ''
2016-05-03 09:18:07.132:WARN:oejs.Response:Committed before 500 org.xml.sax.SAXE
xception|org.eclipse.jetty.io.EofException
2016-05-03 09:18:07.132:WARN:oejs.ServletHandler:ERROR:  /scan/91/data
java.lang.IllegalStateException: Committed
        at org.eclipse.jetty.server.Response.resetBuffer(Response.java:1130)
        at org.eclipse.jetty.server.Response.sendError(Response.java:317)
        at org.csstudio.scan.server.httpd.ScanServlet.doGet(ScanServlet.java:264
)
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:687)
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
        at org.eclipse.equinox.http.servlet.internal.ServletRegistration.service
(ServletRegistration.java:61)
        at org.eclipse.equinox.http.servlet.internal.ProxyServlet.processAlias(P
roxyServlet.java:128)
        at org.eclipse.equinox.http.servlet.internal.ProxyServlet.service(ProxyS
ervlet.java:68)
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
        at org.eclipse.equinox.http.jetty.internal.HttpServerManager$InternalHtt
pServiceServlet.service(HttpServerManager.java:360)
        at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:684
)
        at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java
```

kasemir, 2016-05-12:
The error message is basically:

I tried to send an error message back to the web client, but the Response was already 'committed' to contain something else (likely xml), so I cannot sent the error message back to the client.
As a result, you don't see the original error message.
There have been some changes to the scan server to fix this:
Errors will be logged, without trying to send the error back to the client.
Or log the error, then maybe try to send it back to the client.
In any case, you'd see the actual error in the log.

In your case, all we see is that the error info could not be sent back to the client, and thus we don't know the actual error.

The only clue could come from the line number info, ScanServlet.java:264
Can you compare with your version of the software?
Line 264 in the latest ScanServlet would be
```
           else if ("devices".equalsIgnoreCase(object))
```
so you must be on an earlier version.

berryma4, 2016-05-12:
264: 
``` java
            response.sendError(HttpServletResponse.SC_INTERNAL_SERVER_ERROR, 
```
of a catch
```java
catch (Exception ex)
        {
            response.sendError(HttpServletResponse.SC_INTERNAL_SERVER_ERROR, ex.getMessage());
            Logger.getLogger(getClass().getName()).log(Level.WARNING, "GET /scan failed", ex);
        }
```

Maybe I should just update the person, hope for the best (and get an error message).

kasemir, 2016-05-12:
That's the older code which tried to send the error info to the client, then log.
Sending fails, so we have no idea what the error was.

If you can't update to the latest sources, maybe at least swap those lines to see what the real error is:
````
catch (Exception ex)
{
    // Log error so we can see it
   Logger.getLogger(getClass().getName()).log(Level.WARNING, "GET /scan failed", ex);
   // Try to send error to client, which may fail because client closed,
   // response already committed, ..
   response.sendError(HttpServletResponse.SC_INTERNAL_SERVER_ERROR, ex.getMessage());
}
````

kasemir, 2016-05-12:
As for the actual error, I can only guess. Might be a timeout, which regrettably is at this time not configurable via a preference.

berryma4, 2016-05-12:
I'm guessing it's a timeout, considering I can get it through a browser.

kasemir, 2016-05-12:
In the `ScanClient` class, there's right now only a constant for `default_timeout` and `long_timeout`. Getting the data should probably use the `long_timeout`, and either one should be read from a preference.

closed 2018-06-12 by berryma4