ControlSystemStudio/cs-studio #1790: Drawing polyline always ends with two points

NickeZ, 2016-05-17:
I have noticed that when you end a polyline with double click it generates two points at the end position instead of one. I, for example, cannot find a way to draw a straight line that only contains two points.

kasemir, 2016-05-17:
True.

In the display builder update #1230 the polyline/polygon editor uses the Escape key to end adding points, because distinguishing between click (add) and double-click (don't add, end) is - as you noticed - not so easy.

Your options:
a) Ignore it. The extra point (two points on top of each other) really don't matter.
b) Wait a year, then use the new display builder
c) In BOY, add the line as before, then open the Points dialog (click on the "Points" property of the poly widget) and remove the extra point
d) Update the BOY source code to handle it better

NickeZ, 2016-05-17:
Thank you! I did not know about (c), I was trying in all other different ways to get rid of points..

closed 2016-05-17 by NickeZ