ControlSystemStudio/cs-studio #1792: Getting description from PV into tooltip

NickeZ, 2016-05-17:
I have a use case which I currently can't find an immediate way to solve in CSS: I would like to get the PV DESC field into the mouse over tooltip of the PV.

My current solution is that I write a rule which connects to the DESC field. I think it would be more appropriate if this information was read out at the initialization, like limits and precision, and be a widget property so that the tooltip could use it like this: `$(pv_name)\n$(pv_value)\n$(pv_description)`

Here is the generated script from my rule:

```javascript
importPackage(Packages.org.csstudio.opibuilder.scriptUtil); 
var pvStr0 = PVUtil.getString(pvs[0]);
if(true)
	widget.setPropertyValue("tooltip","$(pv_name)\n$(pv_value)\n" + pvStr0);
else
	widget.setPropertyValue("tooltip","$(pv_name)\n$(pv_value)");
```

btw I also found a minor issue which is that when the default value for the tooltip gets inserted into the script it doesn't have it's newlines escaped and you get the following issue:

```
2016-05-17 11:45:39 ERROR: Failed to register MinRule. 
org.mozilla.javascript.EvaluatorException: unterminated string literal (rule#6)
```

The following is invalid javascript:

```javascript
importPackage(Packages.org.csstudio.opibuilder.scriptUtil); 
var pvStr0 = PVUtil.getString(pvs[0]);
if(true)
	widget.setPropertyValue("tooltip","$(pv_name)\n$(pv_value)\n" + pvStr0);
else
	widget.setPropertyValue("tooltip","$(pv_name)
$(pv_value)");
```

kasemir, 2016-05-17:
I'm sorry, but this is simply not supported by EPICS/Channel Access.
A 'channel' only returns the info in the DBR_CTRL_*  packages. That's what we offer in the PV.

The DESC field of records would be a different channel. If you talk to an IOC, then there is a "fred.DESC" channel available for each channel "fred". But if you talk to another Channel Access server (LabVIEW, pcaspy, ..) then there is no such *.DESC channel.

Similarly, when looking at a motor, people would often like to see the motor's readback position, movement state, .. from the RBV, DMOV, .. fields. But again those are separate Channel Access connections.

If you think you really need that in a tooltip: Are you sure? A tooltip is always somewhat flaky. Need to hold your mouse pointer real still, then hope for the tooltip to appear. Then when you're halfway through reading the content, the tooltip vanishes again. If you think that information matters, add textupdate etc. widgets for it, maybe on a "detail" subscreen.

If you still insist on updating the tooltip: OK, then you will need to add scripts which read the *.DESC, *.RBV, .. whatever other fields of the record you want to see, and only you will now the record type and those fields because it's very hard to determine that automatically. It will make your display run slower, because the scripts updating the tooltips (which most of the time nobody ever sees) will run on the UI thread and slow everything down.

NickeZ, 2016-05-17:
I have found tooltips valuable in my applications and I think it would be nice to actually use the DESC field in the IOC to store this information. DESC is often populated but almost never actually read in my experience. It is also one of the fields that exists in all record types but I agree with you that an IOC is not the only source of PVs.

If I need to have a script that reads out this value it will slow down my screen, that was why I was suggesting that it would be read out at the initialization of the connection instead. I will probably copy/paste the description from the IOC and try to keep my screens up to date with my IOCs instead.

closed 2025-09-10 by NickeZ