ControlSystemStudio/cs-studio #1793: Using PV values in linking container macros

NickeZ, 2016-05-17:
I don't think the following can be done in a nice way so I guess this is a feature request.

## Background
There is a hack that I've been using in some engineering GUIs. The end goal is to have a local (CS-studio) variable that is used in all the PV names. Often it is a prefix, but sometimes it is an infix (like a number).

Imagine that we have a screen which has a linking container and a prefix selector. We want to let the user pick the prefix of the device either via a dropdown menu or a text input box and then use that prefix in the linking container. See following image:

[image]

Currently we need the following to get this behavior:

1. A script attached to the linking container which triggers on the local variable and updates the macros on the linking container.
2. Two rules attached to the linking container which also triggers on the local variable to force reload the OPI.
    * One to clear the opi_file
    * One to set the opi_file back to the default value.

## Request
* I would like to set a PV as part of a macro value.
* If the PV changes it should force a reload of the macros which should force a reload of the linked OPI screen,

kasemir, 2016-05-17:
One idea: Tabs
The tabs allow you to switch between several displays.
Upside: The displays are all running, so switching between them is fast.
Downside: The displays are all running, even those that you don't want to use.

Other Idea: Menu button (or action button)
Your display X.opi has a Menu button (or action button) on top with many actions that all load the very same display X.opi, using the 'replace' target, but different macros.
That's like your drop-down idea, but requires no scripts/rules.

NickeZ, 2016-05-18:
I like the second Idea and I'm quite close. Right now I have an input box and an action button both connected to a "loc://${DID}_prefix". Then the Action button loads the OPI with the macro PREFIX=$(pv_value). But as I'm sure you see this sets the PREFIX macro to "VString[test, 2016/05/17 01:34:21.783]" which is not really what I want...

NickeZ, 2016-05-18:
I'm back to square one. I can implement it with an action button and the following script. But then my first way is more user friendly since the user doesn't have to push an additional action button.

```javascript
importPackage(Packages.org.csstudio.opibuilder.scriptUtil);

var prefix = PVUtil.getString(widget.getPV())

var macrosInput = DataUtil.createMacrosInput(true);
macrosInput.put("PREFIX", prefix)

ScriptUtil.openOPI(widget, "testtest.opi", 1, macrosInput)
```

kasemir, 2016-05-18:
A menu button would work nicely if you have N options, and you know them all
--> Add N actions, each opening the same display but with N different macros.

If you do need to allow entering the "macro value", then yes, you'll need to use a script.
And you'll need to update that script with upcoming versions of CS-Studio.
But your script is only a few lines, so that won't be too bad. Certainly better than trying to create a new type of widget just for this specific purpose.