ControlSystemStudio/cs-studio #1801: Intensity Graph maximum size

lcavalli, 2016-06-13:
Hi,

I have an intensity map to be visualized by an Intensity graph OPI  widget.

The map is 4096 x 4096 ("Data Width" and "Data Height" properties), thus exceeding the maximum size in:
[applications/opibuilder/opibuilder-plugins/org.csstudio.swt.widgets/src/org/csstudio/swt/widgets/figures/IntensityGraphFigure.java](https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/opibuilder/opibuilder-plugins/org.csstudio.swt.widgets/src/org/csstudio/swt/widgets/figures/IntensityGraphFigure.java#L101)

defined as:
`private static final int MAX_ARRAY_SIZE = 10000000;`

Trying a map of 4096 x 2441 elements (below the MAX_ARRAY_SIZE limit) works (slow, but works).

Is there a way to make the maximum map size a configurable parameter?

Thanks for all your work and help,

Luca

kasemir, 2016-06-13:
Sure, you could make this parameter configurable. Feel free to submit a pull request.

In practice, however, I doubt you'll be happy with the results for several reasons.
Like most of the BOY code, the Intensity graph performs its drawing on the UI thread.
Note the comment in the same file where you found the MAX_ARRAY_SIZE:
```
     * Warning: for big image for example 1024*768, it may takes several milliseconds (10-50ms)
     *  to paint the image...
```
You found yourself that 4096 x 2441 elements are already slow. 4096 x 4096 will be even slower, and what's worse is that you're slowing down the UI thread, so all of CSS will be bogged down.

That's the prime motivation behind the new display builder development, see #1230 and presentation from recent EPICS meeting. In there, the Image widget (which is mostly compatible with the Intensity Graph) has no data size limitations (other than running out of JVM memory) and the drawing is performed in a background thread except for the final copy from an in-memory image to the display.

The other issue would be the Channel Access traffic. In most production setups you will have a Channel Access gateway somewhere in your data pipeline, and the CA gateway cannot nicely handle large waveforms. Access to all other PVs will become intermittent because the single-threaded handling of traffic gets bogged down by just one large waveform. The new PVAccess gateway (pva2pva) has much better degradation behavior because of "fair scheduling", as was demonstrated at the recent EPICS meeting at ESS.

So instead of increasing MAX_ARRAY_SIZE, I would downsample the image on the IOC and then only serve a smaller image out for 'life displays'. Mid-term, I would look at serving images via pvAccess, and plan to update to the display builder in about a year.

lcavalli, 2016-07-06:
Closing this issue since increasing array size makes the GUI totally unresponsive. Adopting alternative solutions as suggested.

closed 2016-07-06 by lcavalli