ControlSystemStudio/cs-studio #1809: Databrowser: unable to save a new databrowser plot

berryma4, 2016-06-16:
If a new databrowser plot is created, nothing happens if you click the save icon in the coolbar, or if you go to file->save as.

berryma4, 2016-06-16:
@shroffk was able to reproduce the error.

shroffk, 2016-06-16:
Actually this error was discovered during the CS-Studio training in ESS, it is there in most products.
I will look into this later this week

kasemir, 2016-06-16:
can't reproduce. With the SNS product off the 'master' branch of cs-studio, it looks like this:

* Create new databrowser from toolbar button or CSS/Trends menu
* Save is disabled (no changes and no file name), but File/Save As is available.
* Using File/Save As once
* Add PV, now "Save" becomes enabled on toolbar and in File/Save menu
* Save, and Save options becomes disabled (no changes), but File/Save As still works.

There is the "Misc." option on each *.plt to not save changes. Idea was to not show the "do you want to save?" prompt when people used a plot, zoomed, panned, .. But even with that set the File/Save As remains available.

So what's different in your setups?

shroffk, 2016-06-20:
The problem is due to the absence of the singlesource.rcp  plugin in the product.

After the product.p2 repo is fixed we can try to build the respective products again to check the inclusion of this plugin...the alternative would be to add the o.c.core.utilities.feature to the product.

shroffk, 2016-06-20:
Adding o.c.core.utilites.feature should resolve this issue.

kasemir, 2016-06-20:
The associated UI defaults in o.c.utility.singlesource simply do nothing, which made this harder to figure out. In hindsight, they should log an error.

berryma4, 2016-06-20:
everything is working now (@shroffk and myself verified)

closed 2016-06-20 by berryma4