ControlSystemStudio/cs-studio #1810: Scan Server: exception when wait for completion (put callback) true.

berryma4, 2016-06-17:
If wait for completion is selected, the below exception is written.  If submitted with 'completion=False' then it appears to work as expected.
Scan Server from SNS (v4.1.1 with date Jan. 13, 2016) does not see the error and the scan executes as expected.

```java
2016-06-16 17:00:41.873 CONFIG [Thread 38] org.csstudio.scan.server.internal.ExecutableScan (call) - Executing ID 3 "phyutil 1D Scan" [Memory: Heap: 24.3 / 247.5 MB (9.8 %), Non-Heap: 54.2 MB]
2016-06-16 17:00:41.888 INFO [Thread 38] org.csstudio.scan.server.internal.ExecutableScan (executeWithRetries) - @-1: Wait for devices
2016-06-16 17:00:41.899 INFO [Thread 38] org.csstudio.scan.server.internal.ExecutableScan (executeWithRetries) - @0: phyutil 1D Scan
Comment: phyutil 1D Scan
2016-06-16 17:00:41.905 INFO [Thread 38] org.csstudio.scan.server.internal.ExecutableScan (executeWithRetries) - @1: Set 'V_1:LS1_CA01:CAV1_D1127:PHA_CSET' = 0.0 with completion in 40.0 sec (check for 'V_1:LS1_CA01:CAV1_D1127:PHA_RD' +-0.5)
2016-06-16 17:00:42.913 WARNING [Thread 38] org.csstudio.scan.server.internal.ExecutableScan (executeWithRetries) - Command failed: Set 'V_1:LS1_CA01:CAV1_D1127:PHA_CSET' = 0.0 with completion in 40.0 sec (check for 'V_1:LS1_CA01:CAV1_D1127:PHA_RD' +-0.5)
java.lang.Exception: V_1:LS1_CA01:CAV1_D1127:PHA_CSET (0.0) != 0.0 after completion
        at org.csstudio.scan.server.WriteHelper.perform(WriteHelper.java:132)
        at org.csstudio.scan.commandimpl.SetCommandImpl.execute(SetCommandImpl.java:110)
        at org.csstudio.scan.server.internal.ExecutableScan.executeWithRetries(ExecutableScan.java:623)
        at org.csstudio.scan.server.internal.ExecutableScan.execute(ExecutableScan.java:590)
        at org.csstudio.scan.server.internal.ExecutableScan.execute(ExecutableScan.java:571)
        at org.csstudio.scan.server.internal.ExecutableScan.execute_or_die_trying(ExecutableScan.java:494)
        at org.csstudio.scan.server.internal.ExecutableScan.call(ExecutableScan.java:404)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:745)
Caused by: java.util.concurrent.TimeoutException: Timeout while waiting for V_1:LS1_CA01:CAV1_D1127:PHA_RD (353.81578904832276) = 0.0
        at org.csstudio.scan.condition.NumericValueCondition.await(NumericValueCondition.java:119)
        at org.csstudio.scan.server.WriteHelper.perform(WriteHelper.java:126)
        ... 10 more

2016-06-16 17:00:42.914 WARNING [Thread 38] org.csstudio.scan.server.internal.ExecutableScan (execute_or_die_trying) - Scan phyutil 1D Scan failed
java.lang.Exception: V_1:LS1_CA01:CAV1_D1127:PHA_CSET (0.0) != 0.0 after completion
        at org.csstudio.scan.server.WriteHelper.perform(WriteHelper.java:132)
        at org.csstudio.scan.commandimpl.SetCommandImpl.execute(SetCommandImpl.java:110)
        at org.csstudio.scan.server.internal.ExecutableScan.executeWithRetries(ExecutableScan.java:623)
        at org.csstudio.scan.server.internal.ExecutableScan.execute(ExecutableScan.java:590)
        at org.csstudio.scan.server.internal.ExecutableScan.execute(ExecutableScan.java:571)
        at org.csstudio.scan.server.internal.ExecutableScan.execute_or_die_trying(ExecutableScan.java:494)
        at org.csstudio.scan.server.internal.ExecutableScan.call(ExecutableScan.java:404)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:745)
Caused by: java.util.concurrent.TimeoutException: Timeout while waiting for V_1:LS1_CA01:CAV1_D1127:PHA_RD (353.81578904832276) = 0.0
        at org.csstudio.scan.condition.NumericValueCondition.await(NumericValueCondition.java:119)
        at org.csstudio.scan.server.WriteHelper.perform(WriteHelper.java:126)
        ... 10 more
```

kasemir, 2016-06-17:
I think you're writing to a PV that doesn't support put-callback, i.e. it doesn't support waiting for completion.

`Set 'V_1:LS1_CA01:CAV1_D1127:PHA_CSET' = 0.0 (check for 'V_1:LS1_CA01:CAV1_D1127:PHA_RD' +-0.5 with timeout of 40.0 seconds)` means:

1) Write PHA_CSET = 0.0
2) Wait for PHA_RD = 0.0 +- 0.5 with a timeout of 40 seconds

`Set 'V_1:LS1_CA01:CAV1_D1127:PHA_CSET' = 0.0 with completion in 40.0 sec (check for 'V_1:LS1_CA01:CAV1_D1127:PHA_RD' +-0.5)` means:

1) Write  PHA_CSET = 0.0
2) Wait for the completion callback, allowing 40 seconds
3) Check if  PHA_RD = 0.0 +- 0.5 using a much shorter get-callback timeout, not waiting 40 seconds.

Way back we used to basically apply the timeout  twice:

1) Write  PHA_CSET = 0.0
2) Wait for the completion callback, allowing 40 seconds
3) Wait for PHA_RD = 0.0 +- 0.5 with a timeout of 40 seconds

Then we decided that if you do wait for the completion, the value should match right away, using a get-callback to fetch the latest value.
Operationally, that works much better

... except when you're using the completion callback on a PV that doesn't really support completion.
In that case, the IOC returns the completion callback right away.
The CA client (scan server) can't distinguish between
a) Completion was very quick because we're humming along smoothly
b) IOC returned a fake completion

The latter is what you see:

2016-06-16 17:00:41.905 Set 'V_1:LS1_CA01:CAV1_D1127:PHA_CSET' = 0.0 with completion in 40.0 sec (check for 'V_1:LS1_CA01:CAV1_D1127:PHA_RD' +-0.5)
2016-06-16 17:00:42.913 WARNING  PHA_CSET (0.0) != 0.0 after completion
Caused by: PHA_RD (353.81578904832276) = 0.0
      
--> It did wait for completion, received the (fake) completion within 1 second, and then the value didn't match.

kasemir, 2016-06-17:
The timeout change(applying it twice vs. only for the completion or the readback check) is related to #1301 where it was introduced and #1660 where the error messages were somewhat improved.

dylan171, 2016-06-17:
It had been my understanding that using completion=True was safe even if the channel does not support the put-callback.  As such we had made that the default for our scan helper library.  The default can easily be changed, so this isn't a big issue either way.  But there seems to be a change in behaviour that #1301 could explain.

However, it appears that #1301 was merged Aug 2015 and when I download the scan server from [SNS](https://ics-web.sns.ornl.gov/css/updates/apps/scan-server-4.1.1-linux.gtk.x86_64.zip) which is dated Jan 2016 everything works as I would expect...no exception thrown. Does the SNS version contain the changes from #1301?

kasemir, 2016-06-17:
You're right, the behavior of the earlier binary is a bit unclear.
Maybe it still applied the timeout twice for some reason.

Your "Exception: ..PHA_CSET (0.0) != 0.0 after completion" on the other hand is the expected current behavior when trying to use completion with a channel that doesn't support it.

Sorry, it's not safe to ask for completion when the PV doesn't support it.
Unfortunately there's no way to detect a fake, immediate completion from such channels.
The PyScanClient lib defaults to no completion and we only enable that in the scan settings based on channel name (pattern) like ".*Mot.*" for motors.

dylan171, 2016-06-17:
Thanks for clarifying.  I will update my defaults.

closed 2016-06-17 by dylan171