ControlSystemStudio/cs-studio #1816: Java FX onDebian Jessie

berryma4, 2016-06-29:
```
The program 'Cs-studio' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 26 error_code 2 request_code 155 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

berryma4, 2016-06-29:
java version "1.8.0_91"
Java(TM) SE Runtime Environment (build 1.8.0_91-b14)
Java HotSpot(TM) 64-Bit Server VM (build 25.91-b14, mixed mode)

shroffk, 2016-06-29:
I thought this was resolved with the move to eclipse 4.5.2

Does the old solution of disabling gtk3 not work here?

berryma4, 2016-06-29:
It doesn't seem to:
```
<Desktop >more /usr/lib/cs-studio/cs-studio.ini 
-startup
plugins/org.eclipse.equinox.launcher_1.3.100.v20150511-1540.jar
--launcher.library
plugins/org.eclipse.equinox.launcher.gtk.linux.x86_64_1.1.300.v20150602-1417
-cssTheme
org.eclipse.e4.ui.css.theme.e4_classic
--launcher.GTK_version
2
-vmargs
-Xmx2048m
-Xms128m
-Dorg.eclipse.update.reconcile=false
-Dcom.cosylab.epics.caj.impl.CachedByteBufferAllocator.buffer_size=40000000
-Dosgi.configuration.cascaded=true
-Dosgi.sharedConfiguration.area.readOnly=true
-Dosgi.install.area.readOnly=true
-Dosgi.requiredJavaVersion=1.8
-Dorg.osgi.framework.bundle.parent=ext
-Dosgi.framework.extensions=org.eclipse.fx.osgi
-Dorg.eclipse.swt.internal.gtk.cairoGraphics=false
-Dorg.eclipse.swt.internal.gtk.useCairo=false
```

berryma4, 2016-06-29:
@shroffk @kasemir fun fact, it works the first time cs-studio runs within a new workspace.  If you restart cs-studio, javafx no longer works.

kasemir, 2016-06-29:
That may be a clue. Whatever you're using, the problem only happens when that UI starts up with some persisted info from the last run.

What view do you have open that's based on JavaFX?

I'm using Java 1.8.0_91, but on the older Redhat 6.8.

berryma4, 2016-06-29:
I can repeat it with save restore and log configuration (both javafx)

kasemir, 2016-06-29:
Well, don't see that in the log config for Java 1.8.0_91 on Redhat 6.8.

If you do want to enable the synchronous X:

See the file debug_options.txt in the SNS product:
https://github.com/ControlSystemStudio/org.csstudio.sns/blob/master/plugins/org.csstudio.sns.product/debug_options.txt

If you run with -debug /full/path/to/the/file, and your product is based on org.csstudio.startup, that startup code will use the org.csstudio.startup/debug/async_x=true option.
Everything will run very slow, but on the upside X errors should result in a stack trace that shows the culprit.

berryma4, 2016-06-29:
[image]

kasemir, 2016-06-29:
Is this a quiz ;-) ?

I assume your screenshot is from the native Eclipse launcher. After the JVM exists, it checks for an exit code to know if it should restart Eclipse/CSS, or just exit. In this case, it didn't find such an exit code. Don't remember details, but there was something like a shared memory region set up by the launcher and then written by the Java code, because the "exit code" is more than just a number.

berryma4, 2016-06-29:
I tried to get something from a debugger, with no luck so far:
```
(gdb) break _exit
Breakpoint 1 at 0x7f9786fdc2b0: _exit. (2 locations)
(gdb) break exit
Breakpoint 2 at 0x7f9786f59b60: file exit.c, line 104.
(gdb) break gdk_x_error
Function "gdk_x_error" not defined.
Make breakpoint pending on future shared library load? (y or [n]) y
Breakpoint 3 (gdk_x_error) pending.
(gdb) cont
Continuing.
[New Thread 0x7f977f2f9700 (LWP 31851)]

Breakpoint 2, __GI_exit (status=1) at exit.c:104
104     exit.c: No such file or directory.
(gdb) where
#0  __GI_exit (status=1) at exit.c:104
#1  0x00007f9786f43b4c in __libc_start_main (main=0x401368, argc=3, argv=0x7ffef5876538, init=<optimized out>, 
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffef5876528) at libc-start.c:321
#2  0x00000000004012da in ?? ()
#3  0x00007ffef5876528 in ?? ()
#4  0x000000000000001c in ?? ()
#5  0x0000000000000003 in ?? ()
#6  0x00007ffef5876904 in ?? ()
#7  0x00007ffef5876921 in ?? ()
#8  0x00007ffef5876928 in ?? ()
#9  0x0000000000000000 in ?? ()
(gdb) cont
Continuing.

Breakpoint 1, __GI__exit (status=status@entry=1) at ../sysdeps/unix/sysv/linux/_exit.c:28
28      ../sysdeps/unix/sysv/linux/_exit.c: No such file or directory.
(gdb) where
#0  __GI__exit (status=status@entry=1) at ../sysdeps/unix/sysv/linux/_exit.c:28
#1  0x00007f9786f59aeb in __run_exit_handlers (status=1, listp=<optimized out>, 
    run_list_atexit=run_list_atexit@entry=true) at exit.c:97
#2  0x00007f9786f59b75 in __GI_exit (status=<optimized out>) at exit.c:104
#3  0x00007f9786f43b4c in __libc_start_main (main=0x401368, argc=3, argv=0x7ffef5876538, init=<optimized out>, 
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffef5876528) at libc-start.c:321
#4  0x00000000004012da in ?? ()
#5  0x00007ffef5876528 in ?? ()
#6  0x000000000000001c in ?? ()
#7  0x0000000000000003 in ?? ()
#8  0x00007ffef5876904 in ?? ()
#9  0x00007ffef5876921 in ?? ()
#10 0x00007ffef5876928 in ?? ()
#11 0x0000000000000000 in ?? ()
(gdb) cont
Continuing.                                                                                                             
[Thread 0x7f977f2f9700 (LWP 31851) exited]                                                                              
[Inferior 1 (process 31778) exited with code 01]                                                                        
(gdb) where                                                                                                             
No stack.                                                                                                               
(gdb)   
```

kasemir, 2016-06-29:
The "no exit code" makes a little sense because I assume the JVM crashes, it quits without placing any exit code in the shared mem  (or wherever that was) for the launcher.

Do you attach the debugger to the native lanucher, or to the JVM that's started by the launcher?

berryma4, 2016-06-29:
... funny, I just thought that ... new trace, looks like libswt-pi-gtk-4530.so:
```
Program received signal SIGSEGV, Segmentation fault.
0x00007f30e51c386f in ?? ()
(gdb) handle SIGSEGC nostop
Unrecognized or ambiguous flag word: "SIGSEGC".
(gdb) handle SIGSEGV nostop
Signal        Stop      Print   Pass to program Description
SIGSEGV       No        Yes     Yes             Segmentation fault
(gdb) handle SIGSEGV noprint
Signal        Stop      Print   Pass to program Description
SIGSEGV       No        No      Yes             Segmentation fault
(gdb) cont
Continuing.
[Thread 0x7f3099a35700 (LWP 32174) exited]
[Thread 0x7f309b3e2700 (LWP 32137) exited]
[New Thread 0x7f309b3e2700 (LWP 32237)]
[New Thread 0x7f3099a35700 (LWP 32239)]
[Switching to Thread 0x7f3099a35700 (LWP 32239)]

Breakpoint 1, __GI_exit (status=1) at exit.c:104
104     exit.c: No such file or directory.
(gdb) bt
#0  __GI_exit (status=1) at exit.c:104
#1  0x00007f30db392001 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#2  0x00007f309ab1cca0 in Java_org_eclipse_swt_internal_gtk_OS__1Call ()
   from /user/rea3/.eclipse/2088233279_linux_gtk_x86_64/configuration/org.eclipse.osgi/354/0/.cp/libswt-pi-gtk-4530.so
#3  0x00007f30e5017a14 in ?? ()
#4  0x00007f309b54c268 in ?? ()
#5  0x00007f30f4569510 in ?? ()
#6  0x00007f309b04b7a8 in ?? ()
#7  0x00007f30f4a28000 in ?? ()
#8  0x00007f3099a334b0 in ?? ()
#9  0x00007f30fb6e9039 in MethodCounters::allocate(ClassLoaderData*, Thread*) ()
   from /usr/lib/jvm/java-8-oracle/jre/lib/amd64/server/libjvm.so
#10 0x00007f30e5007d00 in ?? ()
#11 0x0000000080834748 in ?? ()
#12 0x00007f30e5011536 in ?? ()
#13 0x00007f3099a33bb0 in ?? ()
#14 0x00007f309b54a078 in ?? ()
#15 0x00007f3070052790 in ?? ()
#16 0x00007f30e5007d00 in ?? ()
#17 0x00007f30db391f30 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#18 0x0000000000000001 in ?? ()
#19 0x00007f3099a33530 in ?? ()
#20 0x00007f309b04b74b in ?? ()
#21 0x00007f3099a335b8 in ?? ()
#22 0x00007f309b0f1028 in ?? ()
#23 0x0000000000000000 in ?? ()
```

berryma4, 2016-06-29:
with the exit to the "no exit data" pop-up:
```
Program received signal SIGSEGV, Segmentation fault.
0x00007f30e51c386f in ?? ()
(gdb) handle SIGSEGC nostop
Unrecognized or ambiguous flag word: "SIGSEGC".
(gdb) handle SIGSEGV nostop
Signal        Stop      Print   Pass to program Description
SIGSEGV       No        Yes     Yes             Segmentation fault
(gdb) handle SIGSEGV noprint
Signal        Stop      Print   Pass to program Description
SIGSEGV       No        No      Yes             Segmentation fault
(gdb) cont
Continuing.
[Thread 0x7f3099a35700 (LWP 32174) exited]
[Thread 0x7f309b3e2700 (LWP 32137) exited]
[New Thread 0x7f309b3e2700 (LWP 32237)]
[New Thread 0x7f3099a35700 (LWP 32239)]
[Switching to Thread 0x7f3099a35700 (LWP 32239)]

Breakpoint 1, __GI_exit (status=1) at exit.c:104
104     exit.c: No such file or directory.
(gdb) bt
#0  __GI_exit (status=1) at exit.c:104
#1  0x00007f30db392001 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#2  0x00007f309ab1cca0 in Java_org_eclipse_swt_internal_gtk_OS__1Call ()
   from /user/rea3/.eclipse/2088233279_linux_gtk_x86_64/configuration/org.eclipse.osgi/354/0/.cp/libswt-pi-gtk-4530.so
#3  0x00007f30e5017a14 in ?? ()
#4  0x00007f309b54c268 in ?? ()
#5  0x00007f30f4569510 in ?? ()
#6  0x00007f309b04b7a8 in ?? ()
#7  0x00007f30f4a28000 in ?? ()
#8  0x00007f3099a334b0 in ?? ()
#9  0x00007f30fb6e9039 in MethodCounters::allocate(ClassLoaderData*, Thread*) ()
   from /usr/lib/jvm/java-8-oracle/jre/lib/amd64/server/libjvm.so
#10 0x00007f30e5007d00 in ?? ()
#11 0x0000000080834748 in ?? ()
#12 0x00007f30e5011536 in ?? ()
#13 0x00007f3099a33bb0 in ?? ()
#14 0x00007f309b54a078 in ?? ()
#15 0x00007f3070052790 in ?? ()
#16 0x00007f30e5007d00 in ?? ()
#17 0x00007f30db391f30 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#18 0x0000000000000001 in ?? ()
#19 0x00007f3099a33530 in ?? ()
#20 0x00007f309b04b74b in ?? ()
#21 0x00007f3099a335b8 in ?? ()
#22 0x00007f309b0f1028 in ?? ()
#23 0x0000000000000000 in ?? ()
(gdb) cont
Continuing.

Breakpoint 2, __GI__exit (status=status@entry=1) at ../sysdeps/unix/sysv/linux/_exit.c:28
28      ../sysdeps/unix/sysv/linux/_exit.c: No such file or directory.
(gdb) bt
#0  __GI__exit (status=status@entry=1) at ../sysdeps/unix/sysv/linux/_exit.c:28
#1  0x00007f30fbe4faeb in __run_exit_handlers (status=1, listp=<optimized out>, 
    run_list_atexit=run_list_atexit@entry=true) at exit.c:97
#2  0x00007f30fbe4fb75 in __GI_exit (status=<optimized out>) at exit.c:104
#3  0x00007f30db392001 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#4  0x00007f309ab1cca0 in Java_org_eclipse_swt_internal_gtk_OS__1Call ()
   from /user/rea3/.eclipse/2088233279_linux_gtk_x86_64/configuration/org.eclipse.osgi/354/0/.cp/libswt-pi-gtk-4530.so
#5  0x00007f30e5017a14 in ?? ()
#6  0x00007f309b54c268 in ?? ()
#7  0x00007f30f4569510 in ?? ()
#8  0x00007f309b04b7a8 in ?? ()
#9  0x00007f30f4a28000 in ?? ()
#10 0x00007f3099a334b0 in ?? ()
#11 0x00007f30fb6e9039 in MethodCounters::allocate(ClassLoaderData*, Thread*) ()
   from /usr/lib/jvm/java-8-oracle/jre/lib/amd64/server/libjvm.so
#12 0x00007f30e5007d00 in ?? ()
#13 0x0000000080834748 in ?? ()
#14 0x00007f30e5011536 in ?? ()
#15 0x00007f3099a33bb0 in ?? ()
#16 0x00007f309b54a078 in ?? ()
#17 0x00007f3070052790 in ?? ()
#18 0x00007f30e5007d00 in ?? ()
#19 0x00007f30db391f30 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#20 0x0000000000000001 in ?? ()
#21 0x00007f3099a33530 in ?? ()
#22 0x00007f309b04b74b in ?? ()
#23 0x00007f3099a335b8 in ?? ()
#24 0x00007f309b0f1028 in ?? ()
#25 0x0000000000000000 in ?? ()
(gdb) cont
Continuing.
[Thread 0x7f3099a35700 (LWP 32239) exited]
[Thread 0x7f309b3e2700 (LWP 32237) exited]
[Thread 0x7f3099934700 (LWP 32180) exited]
[Thread 0x7f3099b47700 (LWP 32138) exited]
[Thread 0x7f30fc9e6700 (LWP 32079) exited]
[Thread 0x7f30f923a700 (LWP 32080) exited]
[Thread 0x7f30f9139700 (LWP 32081) exited]
[Thread 0x7f30f9038700 (LWP 32082) exited]
[Thread 0x7f30f8f37700 (LWP 32083) exited]
[Thread 0x7f30dbfff700 (LWP 32084) exited]
[Thread 0x7f30dbefe700 (LWP 32085) exited]
[Thread 0x7f30dbdfd700 (LWP 32086) exited]
[Thread 0x7f30dbcfc700 (LWP 32087) exited]
[Thread 0x7f30dbbfb700 (LWP 32088) exited]
[Thread 0x7f30dbafa700 (LWP 32089) exited]
[Thread 0x7f30db9f9700 (LWP 32090) exited]
[Thread 0x7f30db8f8700 (LWP 32091) exited]
[Thread 0x7f30db7f7700 (LWP 32092) exited]
[Thread 0x7f30d04c5700 (LWP 32093) exited]
[Thread 0x7f30d02c3700 (LWP 32095) exited]
[Thread 0x7f309bdff700 (LWP 32097) exited]
[Thread 0x7f309b9e4700 (LWP 32098) exited]
[Thread 0x7f309b6e3700 (LWP 32099) exited]
[Thread 0x7f309b2e1700 (LWP 32101) exited]
[Thread 0x7f30d01c2700 (LWP 32117) exited]
[Thread 0x7f30fc9e8700 (LWP 32078) exited]
[Inferior 1 (process 32078) exited with code 01]
(gdb) bt
No stack.
```

kasemir, 2016-06-29:
Not sure what to learn from the stack traces.

Maybe try to narrow down what causes the crash:
OK with new workspace. Fine.
Crashes on restart if Log config was open. Well, that's already a rather complex display.

Maybe try new workspace, then open the Window, View, Other, Standalone OPI Summary
-> Should be empty, i.e. a very simple JFX display.
Does a restart with that also result in a crash?

One thing I have noticed in the display builder work:
In principle, it should be OK to create some JFX scene, even in a background thread, then create the FXCanvas on the SWT thread and finally call canvas.setScene().
But that doesn't work, I get "Toolkit not initialized".
If I first create the FXCanvas, that seems to initialize some SWT/JFX interoperability stuff. From then on it's OK to create JFX nodes and finally call canvas.setScene().

Maybe that's related to what  you see? On a restart, the code for creating the JFX nodes is invoked too early while the rest of Eclipse is still starting up?

berryma4, 2016-06-29:
I should mention everything links okay:
```
>ldd /user/rea3/.eclipse/2088233279_linux_gtk_x86_64/configuration/org.eclipse.osgi/342/0/.cp/libswt-pi-gtk-4530.so
        linux-vdso.so.1 (0x00007ffcf9142000)
        libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007f0c1e768000)
        libgthread-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007f0c1e566000)
        libXtst.so.6 => /usr/lib/x86_64-linux-gnu/libXtst.so.6 (0x00007f0c1e360000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0c1dfb5000)
        libgdk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0 (0x00007f0c1dcff000)
        libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007f0c1dafb000)
        libpangocairo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007f0c1d8ee000)
        libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f0c1d5ab000)
        libXcomposite.so.1 => /usr/lib/x86_64-linux-gnu/libXcomposite.so.1 (0x00007f0c1d3a8000)
        libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f0c1d1a5000)
        libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f0c1cf9f000)
        libatk-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007f0c1cd78000)
        libcairo.so.2 => /usr/lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f0c1ca5e000)
        libgdk_pixbuf-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007f0c1c83c000)
        libgio-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007f0c1c4c2000)
        libpangoft2-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007f0c1c2ac000)
        libpango-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007f0c1c05e000)
        libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f0c1be0b000)
        libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f0c1bafc000)
        libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f0c1b8bf000)
        libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f0c1b615000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f0c1b314000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f0c1b0f7000)
        libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f0c1aee5000)
        libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f0c1acd5000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f0c1f02d000)
        libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f0c1aacb000)
        libXinerama.so.1 => /usr/lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007f0c1a8c8000)
        libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007f0c1a6be000)
        libXcursor.so.1 => /usr/lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007f0c1a4b3000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0c1a2af000)
        libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f0c1a08d000)
        libpixman-1.so.0 => /usr/lib/x86_64-linux-gnu/libpixman-1.so.0 (0x00007f0c19de0000)
        libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f0c19bb9000)
        libxcb-shm.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007f0c199b5000)
        libxcb-render.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007f0c197ab000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f0c19590000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f0c19388000)
        libselinux.so.1 => /lib/x86_64-linux-gnu/libselinux.so.1 (0x00007f0c19163000)
        libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f0c18f4c000)
        libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f0c18cf5000)
        libthai.so.0 => /usr/lib/x86_64-linux-gnu/libthai.so.0 (0x00007f0c18aec000)
        libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f0c188e4000)
        libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f0c18676000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f0c1844d000)
        libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f0c18249000)
        libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f0c18044000)
        libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f0c17e1e000)
        libdatrie.so.1 => /usr/lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007f0c17c16000)
```

berryma4, 2016-06-30:
Summary so far:
If I start cs-studio locally on the machine, javafx works the first time with a new workspace.
After restarting cs-studio, javafx crashes cs-studio, unless a new workspace is selected.

If I ssh in, and display locally on my windows machine (using Xming for X11), javafx works through restarts, and doesn't run into any issues.

Using a different linux box (Dylan's), we do not see the problem.

This sounds like a graphics card or font issue?  I'm  just not sure why it would work the first time.  What changes in the library linking after a workspace is made?

berryma4, 2016-06-30:
Workspace consists of
```
>find ./ -type f -print0 |xargs -0r file
./.lock:                                                                                                                 empty
./.log:                                                                                                                  ASCII text, with very long lines
./.plugins/org.eclipse.core.resources/.root/.indexes/properties.version:                                                 very short file (no magic)
./.plugins/org.eclipse.core.resources/.root/.indexes/history.version:                                                    very short file (no magic)
./.plugins/org.eclipse.core.resources/.root/4.tree:                                                                      data
./.plugins/org.eclipse.core.resources/.safetable/org.eclipse.core.resources:                                             data
./.plugins/org.eclipse.core.runtime/.settings/org.eclipse.core.resources.prefs:                                          ASCII text
./.plugins/org.eclipse.core.runtime/.settings/org.eclipse.ui.prefs:                                                      ASCII text
./.plugins/org.eclipse.ui.ide/dialog_settings.xml:                                                                       XML document text
./.plugins/org.csstudio.saverestore.ui/dialog_settings.xml:                                                              XML document text
./.plugins/org.eclipse.e4.workbench/workbench.xmi:                                                                       XML document text
```

kasemir, 2016-06-30:
The problem must be caused by something in the X server of that one Linux computer.
It doesn't like some X operation generated by the JavaFX code.
In the past, we've seen that with the SWT "advanced graphics" (anti-aliasing, transparency, ..).
With JavaFX, I've seen a really weird problem if you draw a dashed line over a slider where the value is Double.NaN, when you "rubberband-select" a slider in the editor, where the slider didn't have a value since it's not connected to a PV.

What is the simplest JavaFX-based view where you see this issue?
I think the "Standalone OPI Summary", before you've ever opened any *.opi file, should be just an FXCanvas with a few labels.

berryma4, 2016-06-30:
Ha!  I thought I would check some driver information:
```
<rea3-4:~ >glxinfo
name of display: :0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  155 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  87
  Current serial number in output stream:  88
<rea3-4:~ >glxgears
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  155 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  29
  Current serial number in output stream:  30
```
I'm closing this ticket, this is an IT problem, not CS-Studio.

closed 2016-06-30 by berryma4