ControlSystemStudio/cs-studio #1818: odd selection with grouping containers

berryma4, 2016-07-06:
[image]

@kasemir @shroffk @jbobnar 
If you move your mouse in the purple rectangle up and down on the right side, you will see the action buttons become selected, and can be click.

berryma4, 2016-07-06:
file:
https://drive.google.com/open?id=0B_ikNAN_jxjPaU82RmxmV2NaUkU

berryma4, 2016-07-06:
I do not see this in 4.1.x, but I do in 4.2.x and 4.3.x.

kasemir, 2016-07-06:
We've had a case of buttons that got mysteriously clicked.

Clicking into a Text Entry field would cause a totally different Button to be clicked.
By changing that button style from Custom (draw2D) to Native (SWT button) the problem went away.

We would not figure out why just that one Text Entry field cause that one Button to get activated.
Copied/pasted the Text Entry -> copy would not have the same effect.
Moved the widgets around -> No change.

kasemir, 2016-07-06:
Luca reported a "PV set by clicking unrelated widgets" issue in email:
"The problem is that if I click on some of the Text Update widgets the Boolean Button on top-left part of the GUI switches its status."

kasemir, 2016-07-06:
The groups may be a clue: For Widgets inside a group, their X/Y coordinate is relative to the Group. Draw2D-based widgets seem to confuse the mouse position relative to a group with absolute mouse position within the display.

kasemir, 2016-07-06:
Eric, any recollection how you got the issue?
When I un-group the buttons, the problem is gone.
When I then re-group the buttons, the problem is still gone.
So how to create a display that exhibits the issue in the first place?

kasemir, 2016-07-06:
The 'Classic' buttons are based on  org.csstudio.swt.widgets.figures.ActionButtonFigure.
In there, I can instrument the mouseEntered() method:
```  
      public void mouseEntered(MouseEvent me) {
            System.out.println("Mouse entered @ " + me.x + ", " + me.y);
```

When I now move the mouse over the real button in the issue.opi, I see X coordinates around 30 .. 100.
The same mouseEntered, however, is also triggered when moving over certain areas of the purple group, where the printout shows X coords around 390..460.

Looks like a bug in Draw2D.

I can hack around it in the mouseReleased() method:
```   
     public void mouseReleased(MouseEvent me) {
            if (me.button != 1)
                return;
            if (! containsPoint(me.getLocation()))
            {
                System.out.println("Bogus click");
                return;
            }
```

--> Now the bogus clicks are ignored.

kasemir, 2016-07-06:
This commit works around the issue.
The button will still falsely appear selected, but the click is ignore unless the mouse is really in the button.
Unclear if there's a better way that would fix the underlying mouse coordinate confusion.

berryma4, 2016-07-06:
It seems, if there is a input, or another button, it can only be selected with a right click (because only the left click is the wrong coordinate). 
[image]

berryma4, 2016-07-06:
Same issue with a menu button and a Boolean Button, although not a text input, slider, or spinner.
[image]

kasemir, 2016-07-06:
Can you publish that exact *.opi?

berryma4, 2016-07-06:
https://drive.google.com/open?id=0B_ikNAN_jxjPbDd3T0NFUExBV0E

kasemir, 2016-07-06:
I can confirm that the 
 * Menu Button
 * Bool Button
will erroneously activate from a region within the purple group, because basically
their X/Y relative to the origin of the display matches some mouse location relative
to the upper left corner of the purple group.

I'll try to fix that with a similar ignore-clilck-if-not-really-me hack.

Unsure what you mean regarding the input.
Yes, clicking into the text input would - because of the bug - activate the action button,
but that's fixed on the pull request.
I can still enter text into the text field, no need to right-click.

kasemir, 2016-07-06:
Your example now works for me: [Action|menu|bool] Button all ignore the bogus clicks.

berryma4, 2016-07-06:
.... but this isn't an issue in 4.1.x.  What has changed? (I know the standalone, perspective stuff was added, but ...)

kasemir, 2016-07-06:
is there a difference in swt or draw2d plugin versions?

berryma4, 2016-07-06:
Yes.
Working:
org.eclipse.draw2d_3.9.101.201408150207

Issue:
org.eclipse.draw2d_3.10.1.201508170204

Hilarious, ... you know it is osgi, so I just switched the version and I don't see the problem.

kasemir, 2016-07-06:
Now what?

Fundamentally, I think we want to stay with the current Eclipse release and not use an older draw2d plugin.
We have a workaround by checking the clicked location, so it "works".

Could create a simpler, standalone demo of the Draw2D issue, create a bug report -> I'd rather not spend time on that.

jbobnar, 2016-07-07:
I tried this on Eclipse Neon (which uses a newer draw2d plugin) and it has the same problem.
Looking at the source code difference it is not clear to me what change in draw2d plugin could cause this. @berryma4 Did you replace only the draw2d plugin or anything else as well?

kasemir, 2016-07-07:
I cannot reproduce the problem when I create my own new display.
If I un-group the buttons, then re-group them, the problem is also gone.
So there is something quite strange going on, maybe depending on the order in which Figures are added to a GroupingContainerFigure.
If somebody has the time to create a standalone example, something suitable for submitting a bug report, that could point us to what's really going on.

berryma4, 2016-07-07:
@jbobnar I just added org.eclipse.draw2d specifying version 3.9.101.201408150207 to my product, and the problem _seems_ to go away.  Although, @kasemir is right, I should make a simple non-BOY example to submit to Eclipse ... but I'm still not sure what it is yet.  If it were just a translation issue, it would always be there, as opposed to this temperamental bug.

berryma4, 2016-07-13:
This is a horrible bug, so I am tagging everyone.
@berryma4, @kasemir, @shroffk , @utzeln, @mjclarke01, @charlesHenryPATARD, @ksdj, @xinyuwu, @jpenning, @willrogers, @claudio-rosati

I merged #1819 
But also, I added a p2 with just org.eclipse.draw2d_3.9.101.201408150207:
http://download.controlsystemstudio.org/issue/1818/

If someone wishes, they can add this to their product directly.  For me, I added it as a plugin to my product feature that contains the configuration root files.

berryma4, 2016-07-15:
I should mention, instead of assuming, #1819 will be reverted after draw2d is fixed.  We will leave this ticket open until that time.

willrogers, 2016-10-03:
We have just spotted the same problem in our last beta release and verified Kay's workaround on the latest nightly build.

shroffk, 2016-10-03:
The nsls2 4.3.3 product was modified to explicitly depend on the older version of draw2d (3.9.101) following which the above issue was resolved.
So we seem to have two verified solutions to this.

eddybl, 2020-08-07:
Not sure if this is still unsolved, since the issue seems still to be open. I encountered this issue and did some detailed investigations, here is a copy of an internal bug report I have written earlier:

There is a bug in the linking container widget which makes it possible to generate clicks in other linking container leading to possible unwanted consequences of pushing buttons etc.

Details:
This bug leads to a situation where the clicks of one linking container is mirrored to ALL other linking container preceding the one you are clicking. Preceding means in the .xml file, the placement on the actual OPI panel is not relevant for which containers are effected.

But the placement of the widgets is relevant for this bug to be noticeable, see the attached picture for details. The absolute placement of the containers is relevant for where the "ghost" panel is being placed.

[image]

The clicks are being mirrored in a way where the absolut placement of for example container 1 on the overall panel is available in relative coordinates inside of container 4.
That's also why you can click panel 1 inside of container 3, but cannot affect panel 2 inside of panel 3, since you would need to click more to the right, where the panel is not extended to.
Also the whole panel including borders etc is being available as a "ghost", that's why you can press the buttons in panel 1 if you click between the buttons in panel 2.
The absolute position is again relative to the parent window, so it can also happen inside a grouping container which is not in the top left corner.

With the two panels attached one can easily reproduce the bug, put them to the same folder in CSS and run the panel.opi.
[css-bug-example.zip](https://github.com/ControlSystemStudio/cs-studio/files/5039979/css-bug-example.zip)

smarsching, 2020-09-02:
I did some tests on this and I could verify that this is a bug in the org.eclipse.draw2d plugin. I checked the source  code from https://github.com/eclipse/gef-legacy/ and built various versions, until I could isolate the commit that introduced the bug. It turned out to be commit [e6c829eaf1b36b63b3b67352d64effce781f1477](https://github.com/eclipse/gef-legacy/commit/e6c829eaf1b36b63b3b67352d64effce781f1477).

From the description of the commit, it seems like this change was introduced to fix a bug in which a loop would return too early, returning `null` instead of the element that should have received the mouse event. But it seems like as a side effect, the code now occassionally returns an element which shouldn’t receive the mouse event.

I will have a look at whether I can fix the problem without removing the original bugfix completely, but I am not sure whether any patches for this component are excepted upstream because it seems like it is not maintained any longer (the term “legacy” in the project name and the fact that there have been no commits since 2017 are strong hints)…

smarsching, 2020-09-02:
I found the cause of the bug. In fact, the bug that we see was just hidden by the bug that was fixed in the commit mentioned earlier.

The problem is that the `org.eclipse.draw2d.Figure` class uses singleton instances of `Point` and `Rectangle` in order to avoid allocating short-lived instances.

In my opinion, that’s a classic example of “premature optimization is the root of all evil” (as modern JVMs should be pretty efficient with handling short-lived objects), but I guess the idea was that the whole code isn’t thread-safe anyway, so it was okay to do it. In fact it *would* be okay, if the method wasn’t recursive…

Before the mentioned bugfix, the recursion didn’t matter because the method would always return after the first recursion (and thus the now overwritten data would never be used), but due to the bugfix, `findMouseEventTargetInDescendantsAt` now uses a `Point` that has been overwritten by a recursive call to the same method (through `findMouseEventTargetAt`).

BTW, `findDescendantAtExcluding` would suffer from exactly the same problem, if it didn’t already include a fix for this issue (saving the point’s coordinates on the stack and then using these coordinates directly).

I will make a fork of the `gef-legacy` project, fix the bug in the newest version of the `org.eclipse.draw2d` plugin, and then make a release of that fixed plugin version.

smarsching, 2020-09-03:
I fixed the bug in `org.eclipse.draw2d.Figure` and published an update-site containing the fixed version at https://github.com/smarsching/gef-legacy/releases/tag/snapshot-202009022027, in case anyone else needs it.