ControlSystemStudio/cs-studio #1822: Spinnner widget decimal number format is inconsistent

nickbattam, 2016-07-07:
The spinner widget displays the (locale specific) 1000s separator when using DECIMAL format (e.g. 1,000,0041). Other widgets, like the TextUpdate, that offer DECIMAL format do not.

Looking in the code the number is formatted with a new DecimalFormat() object. For the TextUpdate widget the number is formatted via VTypeHelper using a new DecimalFormat("0").

I would like to update the spinner to use the same formatting as other widgets.

If any site requires the inclusion of the 1000s separator on the spinner I could add it as an option, but that would be a larger code change.

kasemir, 2016-07-07:
I agree with your plan.
In general, we should prefer common, reproducible formats over locales.
CS-Studio displays should look the same no matter where you execute them. Having 1000s separators be added or removed, using either ',' or "'", having decimal separators change from '.' to ',' depending on where you look at a display is not good for control system tools.

berryma4, 2016-07-08:
could you do a pull request to 4.3.x

PR #1823 (2016-07-08): Force spinner to a locale independent, separator free number format
Resolved #1822
PR #1824 (2016-07-08): Force spinner to a locale independent, separator free number format
Resolves #1822 on 4.2.x branch
closed 2016-07-21 by berryma4