ControlSystemStudio/cs-studio #1826: Intermittent failure in diirt unit tests

nickbattam, 2016-07-11:
We have been seeing intermittent failures of the unit tests in diirt, building the 4.3 branch:

[diirt_test_failure.txt](https://github.com/ControlSystemStudio/cs-studio/files/357064/diirt_test_failure.txt)

Rerunning the build on the same hardware straight after both of these failures the unit tests passed successfully.

kasemir, 2016-07-14:
Are the expected values in the test simply too "tight"?

shroffk, 2016-09-08:
Never mind, got it to reproduce...looking into it now.

nickbattam, 2016-09-08:
@shroffk For the record the failures are happening on master as well as the 4.3 branch.

shroffk, 2016-09-08:
@kasemir The idea of the test is to ensure that the simulated updated arrives with the exact timestamp.

To me the problem seemed to be how the replay was calculating the offset...both the ReplayTest and the Replay class create a reference instant now(), when this instance differs by more than a ms the test fails.
So instead I think it is best to calculate the offset for the events from the TimeInterval supplied to the create value. This way we still get values with the exact timestamp as specified in the file.

nickbattam, 2016-09-09:
@shroffk I've run a number of diirt builds with the https://github.com/ControlSystemStudio/diirt/commit/111368054aa84562c7310cf934233753fd3ac323  patch and it resolves the Replay test failure.
I can still reproduce the ErrorHandlingTest failure:

`exceptionInFunction(org.diirt.datasource.sim.ErrorHandlingTest)  Time elapsed: 1.01 sec  <<< FAILURE!`