ControlSystemStudio/cs-studio #1832: getting a view in a specific perspective

berryma4, 2016-07-25:
The ability to move a view to different perspectives.
We've discussed this before, but I thought we should give it a ticket for an official up or down.

willrogers, 2016-09-19:
More specifically:

* Open a view in one perspective
* Change to another perspective
* Get hold of the original view, which has just disappeared on perspective change.

It might be that the 'Quick Access' search widget would do what we want.  You can try this in Eclipse, but it's not there in CS-Studio, at least in our build.

Adding or removing this widget appears to be a dark art.  People complained that it appeared by default in their e4 applications, but now it's not clear how to get it back.

https://bugs.eclipse.org/bugs/show_bug.cgi?id=411821

berryma4, 2016-09-26:
When I say "view", I mean opi view.

berryma4, 2016-09-27:
Yes, I'll add the another though:
 1.  Move a specific opi view to another perspective
     - Open a view in one perspective
     - Change to another perspective
     - Get hold of the original view, which has just disappeared on perspective change.
 2. Make an opi view "pinned" though perspective changes
    - Open view in one perspective
    - "Pin" opi view in position
    - Change to another perspective
    - Pinned opi view is still in its position, in the new perspective.