ControlSystemStudio/cs-studio #1841: Databrowser: Marcos not expanding

berryma4, 2016-07-27:
[image]

If you right click "Open Data Browser" from in BOY; the data and axis (the plot) updates macros correctly, but the Properties View does not.

kasemir, 2016-07-27:
Well, the property view is where you enter the macros, so they can't expand in there..

MichaelRitzert, 2016-07-28:
I just wanted to open a similar issue:
when you want to export to CSV a plot defined with a macro in the PV name, the macro is not expanded.

kasemir, 2016-07-28:
Good point, the export won't actually work unless the macros get resolved, since it can't fetch data from the archive for "$(macro)". Will fix that.

Still, for the property view I see no easy solution since that's where you enter the name, potentially with macros.

shroffk, 2016-07-28:
I agree with Kay, I think the issue is "editing" a databrowser file vs "running" a databrowser file, if you are editing it then you want to see the actual macros but if a user is opening from BOY they want the pv names...

Would it be possible to update the command in the context menu of the databrowser BOY widget such that "open in databrowser" would open a *new* databrowser with all the macro values expanded instead of opening the template file.

kasemir, 2016-07-28:
I'm unsure about the use case. Why is the property panel opened?

When "editing", the property panel is open because you want to adjust things.
When "running", aren't you simply looking at the plot? Zoom/pan is available in the plot, so why open the property panel unless you actually want to "edit"?

berryma4, 2016-07-28:
I think this is just a matter of expected behavior.  When macros are used, the databrowser file is assumed to be a template (because it can't be used without macros to fill in the values).  When the template is opened with macros, the user expects a databrowser instance with all the macros filled.

kasemir, 2016-07-28:
I see.
So right, one solution could be to check at startup: If there are macros, expand all names. But that would then really need to be a new databrowser, where pressing "Save" doesn't overwrite the existing one.
We just started a databrowser version that uses JavaFX for the plot, to allow embedding into the new display builder. Changing the way the macros are handled isn't trivial, and I'd hate to change that twice.
The way macros are handled certainly should change to use the same macro lib for display builder, databrowser, scan. Since the current databrowser isn't broken (plot works, export now works with macros), I'd rather leave the macros being visible in the property view as a minor inconvenience for now.

berryma4, 2016-07-28:
> So right, one solution could be to check at startup: If there are macros, expand all names. But that would then really need to be a new databrowser, where pressing "Save" doesn't overwrite the existing one.

This sounds good to me.  Should we say this will be implemented in the new JavaFX version?

tashasummers, 2018-03-06:
Has there been any work on this issue? I make frequent use of data browser templates for my details screens, and they continue to cause confusion. 

Also, would it be possible to expand a macro in the title? 

[image]

kasemir, 2018-03-06:
The title is resolved in the 'new' databrowser for CS-Studio/RCP as well as phoebus.
The property panel still shows the macros, since that's where you would enter them.

berryma4, 2018-03-06:
@tashasummers did you see the issue in your 4.5 pre release?

tashasummers, 2018-03-06:
@berryma4 yes, this was using 4.5.6

willrogers, 2018-03-07:
Hold on, this isn't fixed in Databrowser 2?

willrogers, 2018-03-07:
Why don't we replace databrowser 2 with databrowser 3?

kasemir, 2018-03-07:
I can do a pull request for db2, if you can test that.
Or everybody updates their product to use databrowser3, and then we can delete db2.

closed 2018-08-13 by berryma4