ControlSystemStudio/cs-studio #1853: Spurious 'dirty editor' in databrowser

willrogers, 2016-08-15:
* open a databrowser editor
* save it to a file
* restart CS-Studio

The editor is now marked as 'dirty' and if I close CS-Studio it prompts me to save the file.  However, I haven't changed anything about the plot and when I do save the file it doesn't change the file's contents.

I find this behaviour confusing.  Am I missing something or is this some kind of bug?

kasemir, 2016-08-15:
When you open the *.plt, does it fetch archived data, and when you look at the properties for the value axes, is it configured as "When archived data arrives: (*) Perform 'stagger'"?

-> As it receives archived data, it updates the Y axes min/max range.
That's considered a configuration change, so the config is marked as dirty.

Ways to work around this:
Configure the Y axes ranges as desired and then use "When archived data arrives: (*) Do nothing".

Or un-check the "Misc." property "Save Changes: [ ]": When that's un-checked, the user or the stagger-when-data-arrives can change anything, the configuration is never considered dirty. That can be useful for production screens that people can open, they can zoom/pan/whatever, but end users are never meant to save such changes in the original file. They can still File/Save As.. under a new name, but the original config is never prompting you to save-on-close because of the dirty config tate.

willrogers, 2016-08-16:
* When you open the `*.plt`, does it fetch archived data?
 * yes
* when you look at the properties for the value axes, is it configured as "When archived data arrives: (*) Perform 'stagger'"?
 * no, it's on Do nothing

When I save, the asterisk disappears.  The `.plt` file (in a Git repo) has not changed.  If I click back on the plot editor, the asterisk comes back.  I can save again and the file still hasn't changed.

Occasionally, when I save, the `.plt` file **has** changed.  In that case it is the Y axis values that have changed.  This can happen if I restart CS-Studio and the plot is redrawn.

kasemir, 2016-08-16:
Just found that there's a quirk with the handling of units:

Open an existing *.plt file, plot opens, PV connects and receives units.
This sends a 'change' event to make the plot update the value axis, now showing the units in the label.
But the same change also tells the editor that something has changed.

This would require adding a separate notification for that change.

Another reason why I see plots here get 'dirty':
One of the archive data sources returns 'channel not found', whereupon it's removed from the list of data sources to prevent asking it again over and over -> config change.

Plus of course any time the user clicks inside the plot and thus "pans" the plot.

Your best option right now if you want to open existing *.plt files and then keep them "clean" is un-checking the  "Misc." property "Save Changes: [ ]" (and then use "Save As" to store that file a last time).

willrogers, 2016-08-16:
Do you think there's something here that we should fix?  You know the insides of databrowser and I don't.

My instinct is that the editor shouldn't be dirty unless there's something that would change in the `.plt` file.  (e.g. there's a difference between _clicking on_ the editor to select it and _clicking and dragging_ the editor which pans and therefore changes the axes).

kasemir, 2016-08-16:
Yes, it would be good to fix these.
The click that turns into a 0..1-pixel pan/drag should be ignored.
The unit quirk might need a separate notification, one that the 'dirty' check can ignore, while the plot still needs to redraw.
Since we just started on a JFX version of the data browser, I'd rather fix it in there.

willrogers, 2016-12-15:
With my new interest in the data browser, I looked at making these changes.  But then I realised that you may have fixed them in the new databrowser so it may be fairly quick to port your changes back to data browser 2.  Is that true?

p.s. It's Christmas soon so you will have a break from me asking questions on Github!

kasemir, 2016-12-15:
Yes, I'll be on a break for Chrismas pretty soon as well.
Still, if you're interested in doing these changes on databrowser2, go ahead and we'll port them into the updated data browser.

closed 2023-10-11 by willrogers