ControlSystemStudio/cs-studio #1854: Databrowser axis behaviour

willrogers, 2016-08-15:
[I find the following behaviour unintuitive but happy to discuss]

If I open a databrowser plot, the value axis is typically on autoscale.  Then if I select a zoom (positive or negative) and try to zoom, the value axis will return to its original scale.

Meanwhile, the time axis is happy to obey my zooming, effectively turning off the 'scrolling' behaviour.

I think what I'd expect is for the value axis also to obey the zoom and to turn off autoscale in order to allow it.

However, I realise that this would be a change of behaviour so I'm interested to find out what other people think @kasemir @shroffk @berryma4.

kasemir, 2016-08-15:
I think you're right: Manually zooming an autoscaled axis should disable the autoscaling.

To avoid any change in behavior for BOY, I'll implement that in the display builder update.

willrogers, 2016-08-16:
> To avoid any change in behavior for BOY, I'll implement that in the display builder update.

Do you mean the widgets?  I mean the databrowser itself, although I understand that might be getting a JavaFX version too.

kasemir, 2016-08-16:
I've updated this in the JavaFX version of the RTPlot. That's used in the XYPlot of the display builder, and also in a JavaFX version of the data browser, which is necessary for having a data browser widget in the display builder. That data browser update to JFX is a larger project, Megan started on it.

willrogers, 2016-08-19:
If we were to change the behaviour, could this be included in 4.4?  

I'd be happy putting it in 4.3 but perhaps it's a bit much to call it a bug fix.

berryma4, 2016-08-19:
I would be happy to do a quick 4.4

mfurseman, 2016-08-25:
So after having a go at implementing this I have two questions.

- Should the change in auto scale behaviour be added to the undo history? Without this when a user zooms the Y axis then clicks the undo button the scale is restored to the previous range, but auto scale remains disabled.
- Should auto scale be disabled when using the pan tool?

willrogers, 2016-08-25:
I think yes when panning.

kasemir, 2016-08-25:
Maybe the UndoableAction for changing the axis range needs to include the autoscale setting. So zooming (and at the same time disabling autoscale) becomes one operation, which will be un-done.

closed 2016-09-14 by berryma4