ControlSystemStudio/cs-studio #1859: upstream changes break build

berryma4, 2016-08-19:
We were discussing this on the core list, and realized there wasn't an issue ticket yet:
https://sourceforge.net/p/cs-studio/mailman/message/35285435/

Basically, the bundle names for pvdata and pvaccess were reverted.  

We had two options; change our code, or ask the EPICS7 people nicely to change the manifest names back.  Since having a bundle name that matches the namspace name is nice, we went for option two, upstream issue tickets were made:

https://github.com/epics-base/pvAccessJava/issues/8
https://github.com/epics-base/pvDataJava/issues/12

Ralph was nice enough to do this rather quickly, so we should make a new version of diirt and fix o.c.platform.lib.epics (fix the mave-osgi-bundles repo).

berryma4, 2016-08-19:
https://github.com/ControlSystemStudio/cs-studio/issues/1857

berryma4, 2016-08-19:
https://github.com/ControlSystemStudio/cs-studio/issues/1855

kasemir, 2016-08-23:
Notes from hangout (Eric, Kunal, Claudio, Kay):

We identified the E4 pvAccessJava and pvDataJava plugins to use (May versions).
We'll remove  o.c.platform.libs.epics(.ui) and instead use the separate jca and caj plugins.
vtype.pv continues with preference-based configuration, it'll have the corresponding code from o.c.platform.libs.epics(.ui).
diirt continues to use its separate XML files.
We work around the class loader issue where JCALibrary.createContext() cannot create the CAJContext by directly calling "new CAJContext()".

Timeline:

Today
Kunal updates diirt to use "new CAJContext()", release new diirt version, update poms.

Tomorrow
Merge #1861, which removes dependency on o.c.platform.libs.epics(.ui), updates vtype.pv

Everybody then needs to update their products:
* Remove o.c.platform.libs.epics(.ui)
* Instead, add the separate JCA and CAJ plugins
* If you want, add o.c.vtype.pv.ui for preference UI and online help
* Update plugin_customization.ini: o.c.platform.libs.epics/* -> o.c.vtype.pv/*

berryma4, 2016-08-23:
@berryma4, @kasemir, @shroffk , @utzeln, @mjclarke01, @charlesHenryPATARD, @ksdj, @xinyuwu, @jpenning, @willrogers @claudio-rosati 
Note to Site Reps, #1861 will break your preferences for caj/jca, please update your product:
 - o.c.platform.libs.epics(.ui) will  be removed
 - Update plugin_customization.ini: o.c.platform.libs.epics/* -> o.c.vtype.pv/*

shroffk, 2016-08-24:
https://github.com/ControlSystemStudio/diirt/commit/8a39eb987a78c1c80697297b81ccbe92cd1cd7a2

Creates the cajContext directly instead of using the jcaLibrary, it continues to use the jcaLibrary to create the other types of context.

kasemir, 2016-08-24:
Products that contained
* org.csstudio.platform.libs.epics
* org.epics.pvdata 4.0.3
* org.epics.pvaccess 4.0.4

need to replace them with
* org.epics.jca 2.3.6
* org.epics.caj 1.1.15
* org.epics.pvDataJava 5.0.3
* org.epics.pvAccessJava 4.1.3

Preferences (and UI) that used to be in org.csstudio.platform.libs.epics(.ui) are now in org.csstudio.vtype.pv(.ui).

berryma4, 2016-08-25:
@kunal do we update core/applications diirt version to 3.1.5?

shroffk, 2016-08-25:
If everyone else is fine with it.

kasemir, 2016-08-25:
Yes, please do. The master build (i.e. when you pull the current master for diirt, maven-osgi-bundles, ...) has been broken for a week.  I suggest we get it back to compile by updating the diirt version and merging this pull request soon.

shroffk, 2016-08-25:
Merge request created for allowing cs-studio master to select diirt version to be used. 
#1878

kasemir, 2016-08-26:
To make the new vtype.pv.ui available, it needs to be added to some feature.
Will try a build where it's in the core.base.feature.

berryma4, 2016-09-20:
build is working, closing ticket

closed 2016-09-20 by berryma4