ControlSystemStudio/cs-studio #1860: BOY - Set line jump for connectors

utzeln, 2016-08-22:
User requirement (CRYO): there is one feature that would increase the “user experience” for HMI developers: automatically implement the gap when two connectors cross. Doing this manually is quite a burden.

I would propose to add new properties to the connector to set the line jump properties:
- line jump style: arc   | gap   | square   | 2 slides  
- add line jump to: none (will remove the line jumps) | horizontal lines   |  vertical lines  

- size of the line jump: 10

utzeln, 2016-08-22:
[image]

berryma4, 2016-08-22:
I like this a lot.  But, we should make sure it would also be supported in the new display builder @kasemir .

kasemir, 2016-08-22:
For the display builder, there's currently no plan from our side to invest development time in connectors. I don't consider it practical to turn the control system display editor into a schematic design tool where crossing wires can connect (with dot marker) or not (with arc | gap | square | ..) with auto-routing. 
Still, if somebody else wants to implement connector support for the display builder, fine.
But note that its model cleanly separates embedded (Linking Container) widgets from their content. The model is completely unaware of the embedded content. The content of an embedded display is a "black box". The embedded window runtime loads the content as a separate model. So there won't be any connectors from the inside of one embedded display to the inside of another embedded display.

utzeln, 2016-08-22:
"So there won't be any connectors from the inside of one embedded display to the inside of another embedded display" - it is unfortunate as it is one of ITER requirement.

kasemir, 2016-08-22:
I don't see a robust way to support it. The content of an embedded display is unknown. It can even change at runtime. So how can a wire connect to a specific widget inside an embedded display?

utzeln, 2016-08-23:
For now, the source and target for a connector is a succession of widget WUID which gives the path <wuid1>_<wuid2>_<wuid3> is using “_” as separator. So if a connector links 2 valves symbols embedded in grouping containers and linking containers, the source and target paths provides the full chain.

One consequence is that if the user delete a grouping container, then the connector cannot be displayed and generates an exception.

Then, our users are developing "super HMI objects" library where a valve symbol is represented by multiple PVs/statuses in a grouping container that can be linked in other displays. Once the library is published, widget WUID cannot be changed nor their hierarchy, otherwise the connectors between these widgets won't work anymore. In this case, we are managing manually the WUID and make sure they do not change.

But we could also propose to recreate a broken connection looking for some part of the path - this will allow the user to adjust the connector if the source and target  are not the good ones, instead of just dropping the connection.

kasemir, 2016-08-23:
As you mention, it's not very practical: Need to manually manage WUIDs, there are exceptions when a WUID is missing.
When you move widget locations, I bet there are cases where your users don't like the way the connectors auto-route. So next you add constraints: Connector number 47 should always stay north of widget 213423, keeping at least 27 pixels to the west of widget 12455 and never cross connector number 97.

Alternatively, how about just using lines? To the end user, they look the same. Yes, if you move a widget, somebody needs to manually move that line end point to again "connect" to the widget, but overall that's likely much easier.

kasemir, 2016-08-23:
You could of course still implement it in a similar way. Since the WUID isn't really unique, you need to edit it manually to make connectors link up, you might as well use the widget name for that purpose. So the connector becomes a link between widget name1/name2/name3 and widget name4/name5/name6, at runtime establishing those widget locations from the embedded containers.
But it'd be up to you to implement this in the display builder, I don't have the resources.