ControlSystemStudio/cs-studio #1862: Add ENG format to TextUpdate widget

nickbattam, 2016-08-22:
The text update widget has FormatType options for exponential, decimal, hex.

EDM includes the additional 'engineering' format (i.e. exponential using powers of 3 only). 

This format is used in a large number of controls in the existing panels we are converting to CSS.

kasemir, 2016-08-22:
I agree that "engineering" is important. The display builder update already had that option, https://github.com/kasemir/org.csstudio.display.builder/blob/master/org.csstudio.display.builder.model/src/org/csstudio/display/builder/model/properties/FormatOption.java
So I'll support your new format option when reading *.opi files.
Thanks!

kasemir, 2016-08-22:
Thanks again for implementing this, and I'll copy your actual formatting code to the display builder.
I'd fallen for the DecimalFormat's built-in support for engineering notation:
"If the maximum number of integer digits is greater than their minimum number and greater than 1, ... .. generate engineering notation". That works for the exponent, but you can then no longer control the precision. Your DIY formatting based on `int power = 3 * (int) Math.floor(log10 / 3);` works better.

closed 2016-08-31 by nickbattam

kasemir, 2016-08-31:
Note that this could break with EPICS V4:

We want the `precision` to do our own formatting into decimal, exponential, engineering representation. EPICS V3 gave us that as a number.

The VType's Display hands us a Format. For now, that's a decimal format where we can call `getMinimumFractionDigits()` to retrieve the precision.

The EPICS V4 Normative Types hand us a display/format that's a string like "%.5f". In the VType Display we can easily wrap that for the default formatter which calls `String.format(that_display_format_string_received_from_V4)`, but it'll then be hard to retrieve the precision for other representations.

kasemir, 2016-08-31:
More specific: PVManager/diirt and also vtype.pv currently create a "PrintfFormat" as the one shown in https://github.com/ControlSystemStudio/diirt/blob/master/support/pva/src/main/java/org/diirt/support/pva/adapters/AlarmTimeDisplayExtractor.java based on the V4 display/format string.
That is a `NumberFormat`, so technically there is a `getMinimumFractionDigits()` method, but it will always return the default (0), regardless of the V4 display/format string.