ControlSystemStudio/cs-studio #1865: "Precision from PV" is not respected in all NumberFormats

nickbattam, 2016-08-22:
Selecting 'precision from PV' is only respected by the Decimal and Default format options in a TextUpdate widget. For Exponential (and variants) formatting a default precision is used.

In attached screenshot: .PREC = 3; Fixed precision = 3 (default precision = 4)
[image]

kasemir, 2016-08-22:
Problem is we don't know the precision from the PV.
PVManager/diirt gave us the VType, and the VType only hands us a `Format`.
Default and Decimal just call that format, which was created by the VType for the precision received from Channel Access.
Hard to get the precision received via Channel Access/PVAccess back out of the Format for the other formatting options.

kasemir, 2016-08-22:
You could try something like this to recover the precision from a VType `value`:

````
if (value instanceof Display)
{
        final NumberFormat format = ((Display) value).getFormat();
        if (format instanceof DecimalFormat)
                    precision = ((DecimalFormat)format).getMaximumFractionDigits();
}
````

That seems to work with all code that calls the diirt `NumberFormats` helper to construct the Format for a precision received from Channel Access.

kasemir, 2016-08-22:
I'm now retrieving the precision from the Format for the display builder, so all number related options can now use either the precision from the PV or a custom precision configured on the widget.

The PV in the snapshot has PREC=2.

[image]

nickbattam, 2016-08-23:
Thanks @kasemir. There is code to that effect in the simplepv.VTypeHelper at the moment, but the recursive nature of the call means (value instanceof Display) is False at the point we need to get the format data for some of the formats.
I just need to restructure the code a little and it should start working.

closed 2016-08-31 by nickbattam