ControlSystemStudio/cs-studio #1866: Beast: unit test failure

berryma4, 2016-08-22:
I've been getting this unit test failure (on 4.3 and master)

``` java
Test set: org.csstudio.alarm.beast.client.GUIUpdateThrottleUnitTest
-------------------------------------------------------------------------------
Tests run: 1, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 0.394 sec <<< FAILURE! - in org.csstudio.alarm.beast.client.GUIUpdateThrottleUnitTest
testThrottle(org.csstudio.alarm.beast.client.GUIUpdateThrottleUnitTest)  Time elapsed: 0.39 sec  <<< FAILURE!
java.lang.AssertionError: expected:<1> but was:<0>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:645)
	at org.junit.Assert.assertEquals(Assert.java:631)
	at org.csstudio.alarm.beast.client.GUIUpdateThrottleUnitTest.testThrottle(GUIUpdateThrottleUnitTest.java:47)
```

kasemir, 2016-08-22:
I assume not every time, but occasionally?

This test is based on timing.
An event is supposed to be delayed by 100 millisec, then we expect it after twice that time.

Maybe that's great for a demo, but not suitable as a test, because timing is hard to predict on a build server?

berryma4, 2016-08-22:
Works for me.
Thank you!

PR #1867 (2016-08-22): alarm test: Change GUIUpdateThrottle test into demo
Fixes #1866
closed 2016-08-22 by berryma4