ControlSystemStudio/cs-studio #1880: Update saverestore/masar to new pvAccessJava

kasemir, 2016-08-25:
Another side effect of the V4 changes, #1859:
````
Missing requirement: org.csstudio.saverestore.masar 1.0.0.qualifier requires 'bundle org.epics.pvaccess 4.0.0' but it could not be found
````
Needs to use org.epics.pvAccessJava 4.1.3 instead of org.epics.pvaccess 3.0.0.
May also need to update reference to pvdata/pvDataJava

kasemir, 2016-08-25:
@berryma4 Eric, I think all that's required is change `org.csstudio.saverestore.masar/META-INF/MANIFEST.MF` to pvAccessJava and pvData, but I have no way to check the result.

berryma4, 2016-08-25:
Thank you so much.  I can test it tomorrow.  This should be fine.

kasemir, 2016-08-26:
I've grepped for org.epics.pvaccess in all the other MANIFEST.MF, and saverestore appears to be the last one that we have to fix.

lcavalli, 2016-08-26:
I can confirm that the patch fixes the building problem.
Thanks.

kasemir, 2016-08-26:
Excellent!
Merge?

closed 2016-08-26 by berryma4