ControlSystemStudio/cs-studio #1882: Add 'get labels from PV' property to BoolButton

nickbattam, 2016-08-26:
For a BoolButton attached to a BO record, it would be useful to take the 'on' and 'off' text from the ONAM and ZNAM fields of the PV. At the moment, the button has static text for the label when 'on' and 'off' ('On Label' and 'Off Label'). The string value of the PV appears in the button's tooltip.

Dynamic labels are available in EDM, and are used extensively in panels we are converting.

I would like to add a 'get labels from PV' property to the BoolButton, defaulting to 'False' to maintain existing behaviour. There is an 'Show Boolean Label' property at the moment which hides all labels, leaving an unmarked toggle button, I don't believe this behaviour needs to change.

The only implementation detail is deciding the 'relative priority' of the two label sources if On or Off labels are defined and 'get labels from PV' is selected:
1- if the explicit labels override the PV string it would allow the custom override of On or Off text for a particular button
2- if PV strings overrides explicit labels this is a more 'obvious' interpretation of the boolean flag

I think (2) is clearer.

kasemir, 2016-08-26:
Good idea, go for it. Should have been there from the start.

As for precedence, there are other widgets which have a "get limits from PV" option as well as individual high, low, ... limit properties. Or "use precision from PV" and a separate precision property. In those cases, the custom properties are numbers, so there's no good way to _not_ enter anything. You get a zero as default.
In those cases, selecting "get .. from PV" always takes precedence over what you enter in the other fields.
So I think it would make sense to treat this the same way:
You can always enter 'On Label' and 'Off Label'. If you select 'get labels from PV', they're ignored.

willrogers, 2016-09-15:
OK - I've finished off what Nick was doing on this one.

One omission remains: it is not a dynamic property.  If you set a script to change the property value after creation, the labels on the buttons don't change.

I was following the Menu Button and its 'Actions from PV' property for this, and it has the same behaviour.

I'm not sure if this is a minor bug, or if a lot of properties in BOY are like this.

berryma4, 2016-09-20:
Could we have a pull request for master too?

closed 2016-09-26 by berryma4